esphome/feature-requests #2557: Add support for ESP-IDF's CONFIG_LWIP_DHCP_RESTORE_LAST_IP

dbaarda, 2024-01-12:
**Describe the problem you have/What new integration you would like**

Currently when using DHCP and deep-sleep, a full DHCP IP negotiation is required at every wakeup, which requires time and power. The current best-practice recommendation is to use static IPs to avoid this, but it makes managing all the devices a pain.

In theory this is not required, as the DHCP lease has a timeout that is typically 1h or more, and the DHCP lease could be saved in RTC memory and restored on wakeup. This would mean the DHCP lease would only have to be re-negotiated when the lease expires, avoiding lots of DHCP wifi activity on every wakeup. This is similar to how `fast_connect: true` saves the last connected access point and avoids doing a full scan every wakeup.

Many deep-sleep applications will sleep and wakeup every 1m, so saving and re-using a 1h DHCP lease will save about 59 dhcp IP negotiations every hour.

It turns out ESP-IDF already has support for this with a CONFIG_LWIP_DHCP_RESTORE_LAST_IP option. See the following docs for details;

https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-lwip-dhcp-restore-last-ip

**Please describe your use case for this integration and alternatives you've tried:**

Static IP's is an alternative, but it's annoying to manage.

**Additional context**

It's possible there already exists a way to enable this ESP-IDF config option for ESPHome, but I've not been able to find any documentation explaining how. Perhaps all that is needed here is docs explaining how to do it.

dbaarda, 2024-08-15:
Should I have filed this as a bug to get more attention?

candrews, 2026-09-05:
This option can already be set:

```yaml
esp32:
  framework:
    type: esp-idf
    sdkconfig_options: 
      CONFIG_LWIP_DHCP_RESTORE_LAST_IP: y
```

Per https://github.com/esphome/esphome/pull/17073, esphome is working on "a planned DHCP-lease cache (to speed up reconnection for sleep/wake devices)" but that seems to be exactly what  `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` already is.
 
1. Should esphome set `CONFIG_LWIP_DHCP_RESTORE_LAST_IP`?
2. Why would esphome create its own dhcp lease cache when the sdk offers it via `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` already?

🤞 @bootc @clydebarrow or someone else similarly knowledgeable can chime in. Thank you in advance to whoever provides some insight.

dbaarda, 2026-09-06:
> This option can already be set:

That is awesome! Thanks heaps for the tip on how to set it. I'm going to use this on all my devices from now on.

bootc, 2026-09-06:
> [!NOTE]
> This reply was drafted by Claude (the AI assistant) on my behalf, and reviewed by me before posting.

A couple of clarifications first, since I was tagged.

I'm a drive-by contributor here, not part of the ESPHome project — so "esphome is working on…" isn't quite right. #17073 (the RTC preferences backend) was a change I contributed, and the DHCP lease cache is a personal follow-up I've since opened as **esphome/esphome#17767**. Neither is an official roadmap item; whether the lease cache is something the project wants to carry is entirely up to the maintainers.

On the substance, `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` and #17767 are not the same thing — they sit at different points on a correctness-vs-power trade-off. Having read the ESP-IDF source to be sure:

- **Storage:** `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` saves the last IP in **NVS (flash)**, not RTC memory as the first post assumed — namespace `"dhcp_state"`, written on every DHCP bind (see [`components/lwip/port/esp32xx/netif/dhcp_state.c`](https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32xx/netif/dhcp_state.c)). It survives power loss, but a lease changes on every renewal, so it writes flash each time.
- **Mechanism:** it does **not** skip DHCP. On reconnect it enters lwIP's REBOOTING state and sends a DHCPREQUEST for the old address (RFC 2131 INIT-REBOOT: requested-IP option, no server-id); the address is only bound after the server ACKs, and a NAK falls back to a full DISCOVER.

So it's a protocol-correct fast path: one round trip instead of the full four-way handshake, with the server confirming before the address is used.

#17767 takes the more aggressive route: apply the cached lease as a static IP and run **no** DHCP at all while the lease is provably still valid (usable in ~140 ms vs ~2.3 s for full DHCP), gated on the reset reason, lease age/T1, a safety margin and an optional `max_age`. That buys the biggest deep-sleep saving and works even if the DHCP server is briefly unavailable at wake, at the cost of a small, bounded stale-address risk that `RESTORE_LAST_IP` avoids by always asking the server.

| | `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` | #17767 (trust & skip) |
|---|---|---|
| DHCP on wake | One round trip (REQUEST→ACK), server must answer | None — cached lease applied as static IP |
| Time to usable IP | REQUEST/ACK, waits on server | ~140 ms, no exchange |
| Stale-IP risk | None — server ACK/NAK confirms before binding | Small, bounded — reset-reason gate, lease-age/T1 checks, margin, `max_age` |
| Storage | NVS/flash, written every bind (wear); survives power loss | RTC by default (no wear), flash opt-in |
| Lease-expiry awareness | None (relies on server NAK) | Tracks lease/T1, validates age locally |
| Config | SDK flag only (`sdkconfig_options`) | `lease_cache:` with storage / `max_age`, multi-network match |

Answering the two questions directly:

1. **Should ESPHome set `CONFIG_LWIP_DHCP_RESTORE_LAST_IP`?** That's a maintainer call. It's a reasonable, zero-code option and you can already enable it yourself exactly as shown above. The trade-offs are flash wear and that it still runs a DHCP exchange on every wake.
2. **Why a separate lease cache?** For the no-exchange-at-all power/latency saving, RTC storage that avoids flash wear, lease-age awareness, and an ESPHome-level config surface. It's a different trade-off, not a duplicate — and the protocol-correct INIT-REBOOT path (which would close most of the correctness gap) is noted as a follow-up in that PR.

If `RESTORE_LAST_IP` already meets your needs, it's a perfectly good answer today.

dbaarda, 2026-09-24:
@bootc : thanks heaps for the clarification, and #17767 does sound much better for most applications. I do have one question though. For `CONFIG_LWIP_DHCP_RESTORE_LAST_IP`, does that write to flash for every `REQUEST/ACK` on wakeup, or only if the request is `NAK`ed and a new IP issued?

If it's written for every `REQUEST/ACK` even when the IP doesn't change, then that's pretty nasty on flash wear if you are doing sleep/wake every 1min. If it's only written when the IP changes, then provided you have long duration IP leases (1day+) then it's probably not too bad.

dbaarda, 2026-09-24:
I had a look at the code and it looks like it only calls `dhcp_ip_addr_store()` if the IP address has changed here;

https://github.com/espressif/esp-idf/blob/188e3e55bb20505b5fa9dc49ee390bf42c58b4ee/components/esp_netif/lwip/esp_netif_lwip.c#L1548

So provided the DHCP leases are 1d+ and/or never change the flash wear shouldn't be a problem. I have my dhcp server (dnsmasq) setup to always assign the same IP address to each ESPHome device, so it should only get written once until I change the IP address->device mapping in my dhcp config.

However, #17767 honouring the lease time and avoiding the DHCP `REQUEST/ACK` cycle every 1min wakeup would still be better.