esphome/feature-requests #2558: Integration sensor reset to value

hpirila, 2024-01-14:
**Describe the problem you have/What new integration you would like**

Current functionality can only reset to value zero.

## sensor.integration.reset Action

This [Action](https://esphome.io/guides/automations#config-action) allows you to reset the value of the integration sensor to zero. For example this can be used to reset the integration sensor to zero at midnight with a time-based automation.

```
on_...:
  - sensor.integration.reset:  my_integration_sensor
```

In new functionality I would like to optionally set value after reset.

## sensor.integration.reset Action

This [Action](https://esphome.io/guides/automations#config-action) allows you to reset the value of the integration sensor to zero or optionally specific value. For example this can be used to reset the integration sensor to value 105.

```
on_...:
  - sensor.integration.reset:  my_integration_sensor
     value: 105
```

**Please describe your use case for this integration and alternatives you've tried:**

This could be useful to set the stored value if hardware needs replacement. Currently I have to do some tricks to increase integration sensor to specific value over time in order to set correct value. 

**Additional context**

My main interest is really how to set the stored values to specific value when hardware needs to be replaced. Please advice possible methods.

adamnet01, 2024-10-24:
Has anyone managed to solve this?

jurgen2005, 2025-01-04:
up

jurgen2005, 2025-01-20:
is this possible somehow?

mrkvo, 2025-01-21:
Hi, yes.
```
external_components:
  - source:
      type: git
      url: https://github.com/mrkvo/esphome.git
      ref: feature-integration-reset-to-value
    components: [ integration ]

```
Usage in lambda:
```
        then:
          - lambda: |-
              id(IntegrationSensor).reset_to_value(0);
```

Enjoy.