esphome/feature-requests #2561: Add option to force heat/cool on switch on in thermostat

CzBiX, 2024-01-15:
**Describe the problem you have/What new integration you would like**

when the current temperature is greater than target - deadband but less than target, switch to heat mode will not trigger heat.

**Please describe your use case for this integration and alternatives you've tried:**

Say I set the target temperature to 30 degrees and the deadband is 5 degrees. the heater mode is off and current temp is 28 degrees. Theoretically, it will only start to heat if the temperature is below 25 degrees.
What I want is when I set it to heat mode, the thermostat still start heating until the target temp. Even though it's in deadband range.

**Additional context**

An option like `heat_deadband` could be added, it replaces `heat_deadband` only when thermostat is turned on.

ssieb, 2024-01-16:
Why do you have such a large deadband?

nagyrobi, 2024-01-16:
Yeah, deadband should be less than 1 degree... Half degree maybe

CzBiX, 2024-01-16:
Thanks for the reply. There are two reasons that I want a such options.
I am using this for controlling the water heater for showering. I may not take a shower as soon as it reaches the target temperature. It doesn't need to heat up to the target temperature as often during this time. As soon as I am going to the shower, turning on the heat will reach the target temp in a relatively short time. In the meantime I can do the shower prep. so the deadband is big as 5 degrees.
When I started using hot water, there was a very, very large delay in the temp sensor. The water temp doesn't start to drop until I've already been in the shower for 10 minutes. And by this time there isn't much hot water left, at which point it's too late to reheat it. So I used to force it to start heating up when I started the shower.
I know I can set the target water temp higher as a workaround. But I don't need it that high except when I want forced heat. This way seems very twisted to me.

nagyrobi, 2024-01-16:
Rise the warm water temperature a bit, so you mix it with cold water in your shower. That way less warm water will be consumed -> it will last longer (starts temp decrease slower).

For best results what you need is a thermostatic shower mixer valve in your shower cabin, where you set the desired water temperature in the cabin, and the valve automatically keeps temperature constant by mixing the amount of cold and warm water based on their current temperatures.  That way, you can set the temperature of your water a bit higher, with a wider deadband.

CzBiX, 2024-01-18:
@nagyrobi thank you, i got it.

closed 2024-01-18 by CzBiX