esphome/feature-requests #2562: Add support for Stereo Microphones in addition to select Left/Right

MStapelfeldt, 2024-01-15:
**Describe the problem you have/What new integration you would like**
While ESPHome works great now with I2S microphones at left OR right channel, it is not supporting stereo microphones:
```yaml
microphone:
  - platform: i2s_audio
    channel: stereo
```
> channel (Optional, enum): The channel of the microphone. One of left or right. Defaults to right.

**Please describe your use case for this integration and alternatives you've tried:**
At issue #2430 and #2258 request for mic-arrays is already requested and both, [esp32-korvo-v1.1](https://github.com/esphome/feature-requests/issues/2430#top) and [i2s microphone array (6+1)](https://www.dfrobot.com/product-1976.html) uses Left and right channel setups (Korvo ES7210 has L/R Channel description in [Section 5](http://www.everest-semi.com/pdf/ES7210%20PB.pdf).
Besides this (almost) every I2S/MEMS mic has option to select L/R by SEL Pin tied to GND or VDD. With stereo integration, it will be easy setup device with 2 mics pointing 90° angle to catch better voice commants on either direction. 

**Additional context**
Stereo should be already be possible by [espressif idf](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html) description:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#std-tx-mode
> std_cfg.slot_cfg.slot_mode = I2S_SLOT_MODE_STEREO / I2S_SLOT_MODE_MONO; // Default is stereo
(many more details can be found, just search stereo)

ALX-TH, 2024-01-22:
Would be a great to have such feature. +1

Hedda, 2024-08-19:
FYI, stumbled on this new "***voice-kit***" GitHub repository where ESPHome and Home Home Assistant developers from Nabu Casa are developing several new or improved/enhanced audio and media related components, including extended I2S audio for i2c audio microphone and  i2s audio speaker support, including using audio processor (audio DSP hardware) as well as a new native media player with support for FLAC, WAV/PCM, and more advanced resampling, etc. for the upcoming voicekit hardware reference project from Nabu Casa:

- https://github.com/esphome/voice-kit/
  - https://github.com/esphome/voice-kit/tree/dev/esphome/components
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/nabu
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/media_player.py
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/nabu_media_player.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_pipeline.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_reader.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_decoder.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/resampler.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_mixer.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_resampler.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/flac_decoder.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/wav_decoder.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/mp3_decoder.cpp
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio
      - https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio/speaker
      - https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio/microphone
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/media_player
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/voice_kit
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/audio_audio_processor
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/voice_assistant
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/speaker
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/nabu_microphone
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/microphone

Note that I'm only paraphrasing but one of the representatives that is working on that "***voice-kit***"project more or less wrote there that while ESPHome and Home Assistant voice developers from Nabu Casa are now focusing to work on voice assistant features they are still figuring all this around making use of external audio processors and while they are currently only testing the XMOS xCORE chip as a candidate for an ESPHome-based voice-kit reference hardware design for official Home Assistant Voice Assistant development kit they also plan to work on "audio processor" component for ESPHome with hardware-independent architecture that will not be reliant on specific hardware configurations or dependent specifically on the XMOS xCORE DSP chip but instead allow others to add support for additional DSPs as audio processors (i.e. sound co-processors) in the future, (plus the fact that they will make it so that all the I2S settings and pins are still configurable in YAML, meaning that it should at least be possible add support for DSP types to the  "audio processor" component if they work similar to XMOS xCORE DSP chips, as well as different board designs that uses other I2S settings and pins).  That representative also wrote; "_**we will add all the code to the base ESPHome project once things are stable and working well**_".m and noted that ESPHome and Home Assistant / Nabu Casa developers are right now moving very fast and breaking things as they go so working on code for the new voice-kit related components for ESPHome in a separate repository on GitHub here: 

* - https://github.com/esphome/voice-kit/

Anyway, they already added several new experimental features and functions or improvements/enhancements to ESPHome, including a new `nabu_microphone` component that can add better support different microphone channel audio streams.

* https://github.com/esphome/voice-kit/tree/dev/esphome/components/nabu_microphone

and 

* https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio

plus 

  * https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio/microphone

Again, still very early in development so addtional components and more development or changes will sure to come there soon.