esphome/feature-requests #2566: Board Request: Lilygo T-Internet-POE

Lgeorge-V, 2024-01-19:
**Describe the problem you have/What new integration you would like**

I would like the Lilygo T-Internet-POE board to get direct support in the esphome environment. 
[image]

**Please describe your use case for this integration and alternatives you've tried:**

This ESP32 board supports both being powered by POE and using ethernet.

**Additional context**

Here is a link to the exact board I am talking about.
https://www.lilygo.cc/products/t-internet-poe

nagyrobi, 2024-01-20:
Did you look at the docs?
https://esphome.io/components/ethernet

Lgeorge-V, 2024-01-20:
Yes, I looked at the docs, however having this board as a board type in the config of esphome would be nice.

nagyrobi, 2024-01-20:
Perhaps you didn't notice, but in the docs the 3rd configuration example shows the specific config for this board.

closed 2024-01-20 by nagyrobi