esphome/feature-requests #2567: Grow Fingerprint Reader: Possibility to change security level (FRR)

Caros2017, 2024-01-20:
**Describe the problem you have/What new integration you would like**

The fingerprint Grow implementation has a representation of the current security level:
`
security_level: The integer representation of the currently configured security level of the reader. Higher security levels reduce the false acceptance rate (FAR) at the expense of increasing the false rejection rate (FRR). Range is 1 (lowest) to 5 (highest).
`

ESPHome uses the default value `3`   

**Please describe your use case for this integration and alternatives you've tried:**

I would like to be able to set the FRR of the Grow Fingerprint Reader in the range from 1 to 5. Then I can try to make it even more secure.

**Additional context**

It’s described in the adafruit library referenced from the ESPHome doc page. But I don't have the knowledge to implement this myself.

alexborro, 2024-03-07:
Hi @Caros2017 , yes, it is possible to implement this feature. As soon as I get my PR https://github.com/esphome/esphome/pull/6116 merged, I will implement it.

Jenik5, 2025-02-20:
Is there any development?