esphome/feature-requests #2569: Add support to SAGEMCOM XT211 smart meter installed by Groupe-E in Switzerland

binksjj, 2024-01-20:
**Describe the problem you have/What new integration you would like**

The  DSMR version installed in the smart meter is version 5. But the following data are not provided:
Energy Consumed Tariff 1: Unknown
Energy Consumed Tariff 2: Unknown
Energy Produced Tariff 1: Unknown
Energy Produced Tariff 2: Unknown
Power Consumed: Unknown

Please add the following data:
1.8.0 Energy Consumed Total (1.8.1 + 1.8.2)
1.8.1 Energy Consumed Tariff 1
1.8.2 Energy Consumed Tariff 2
2.8.0 Energy Produced Total (2.8.1 + 2.8.2)
2.8.1 Energy Produced Tariff 1
2.8.2 Energy Produced Tariff 2
16.7.0 Power instantaneous Consumed
**Please describe your use case for this integration and alternatives you've tried:**

The DSMR Component is not compatible withe the DSMR version installed in the SAGEMCOM XT211 smart meter installed by Groupe-E in Switzerland.
**Additional context**

razowski, 2024-01-27:
Hello @binksjj,

I think the P1 Port isn't activated by Group-e by default. I've bought another solution (gplug) and same result even if the XT211 is reported as "compatible". I've sent an email to Group-e to ask to activate the P1 Port. I hope they will do it...

Regards,
Razowski

binksjj, 2024-01-28:
> Hello @binksjj,
> 
> I think the P1 Port isn't activated by Group-e by default. I've bought another solution (gplug) and same result even if the XT211 is reported as "compatible". I've sent an email to Group-e to ask to activate the P1 Port. I hope they will do it...
> 
> Regards, Razowski

Hello @razowski ,

Yes, the P1 Port isn’t activate by default. You should ask Groupe-E. In a few months, you will be able to activate it online, through your Groupe-E account. 

My P1 Port is activated since January 8th. Obviously, the DSMR version installed by Groupe-E is not totally compatible with v.5. The port is powered but some data are not available.

I use SlimmeLezer+. But I don’t know how to manually add the missing OBIS code.

Best regards.

razowski, 2024-01-30:
Thanks for the information about Group-e... Waiting on them. I will try another P1 module and let you know the result (in a few days)

razowski, 2024-03-05:
I've give it a try with another P1 module but same result since the DSMR is special as usual with our favorite provider of electricity...
So +1 for the request...

thayoz, 2024-03-18:
> Hello @razowski ,
>
> Yes, the P1 Port isn’t activate by default. You should ask Groupe-E. In a few months, you will be able to activate it online, through your Groupe-E account.
>
> My P1 Port is [...]

Hello @binksjj,

Who have you contacted to activate le P1 one the new Group-E smart meter ? I’ve tried over the hotline with no luck …

thanks

Leefex, 2024-03-27:
Hi all,
I'm doing the same research. Since few days, I'm having an XT211 from Groupe E, and I'm trying to get datas on my Raspberry. For this purpose I have ordered a P1 monitoring cable (rj12 to USB), which is supposed to be compatible with the Sagemcom smart reader. But I have not been able to get values from the smart reader.

# udevadm info --a --name /dev/ttyUSB3
Unknown device "/dev/ttyUSB3": No such file or directory

or

# cu -l /dev/ttyUSB3 -s 115200 --parity=none -E -q
cu: open (/dev/ttyUSB3): No such file or directory

Have any one succeed  with the P1 port activation?

binksjj, 2024-03-27:
> > > Hello @binksjj,
> > > I think the P1 Port isn't activated by Group-e by default. I've bought another solution (gplug) and same result even if the XT211 is reported as "compatible". I've sent an email to Group-e to ask to activate the P1 Port. I hope they will do it...
> > > Regards, Razowski
> > 
> > 
> > Hello @razowski ,
> > Yes, the P1 Port isn’t activate by default. You should ask Groupe-E. In a few months, you will be able to activate it online, through your Groupe-E account.
> > My P1 Port is activated since January 8th. Obviously, the DSMR version installed by Groupe-E is not totally compatible with v.5. The port is powered but some data are not available.
> > I use SlimmeLezer+. But I don’t know how to manually add the missing OBIS code.
> > Best regards.
> 
> Hello @binksjj,
> 
> Who have you contacted to activate the P1 interface of the new Groupe-E smart meter ? I’ve tried over their hotline with no luck …
> 
> thanks

Hello @razowski,

To activate the P1 port, I sent several emails to customer service, made several calls over several months and I even wrote a registered letter to the management of the groupe-e. Finally, I received the following form to fill out.
[Groupe E_Demande d'activation Port 1.pdf](https://github.com/esphome/feature-requests/files/14779876/Groupe.E_Demande.d.activation.Port.1.pdf)

binksjj, 2024-03-27:
> Hi all, I'm doing the same research. Since few days, I'm having an XT211 from Groupe E, and I'm trying to get datas on my Raspberry. For this purpose I have ordered a P1 monitoring cable (rj11 to USB), which is supposed to be compatible with the Sagemcom smart reader. But I have not been able to get values from the smart reader.
> 
> $ udevadm info --a --name /dev/ttyUSB3 Unknown device "/dev/ttyUSB3": No such file or directory
> 
> or
> 
> $ cu -l /dev/ttyUSB3 -s 115200 --parity=none -E -q cu: open (/dev/ttyUSB3): No such file or directory
> 
> Have any one succeed with the P1 port activation?

Hello @Leefex, look at my response in the previous message.

Leefex, 2024-03-27:
many thanks for your help, the form is filled up and is on the way.

jacoch, 2024-04-24:
Groupe E just activated my port and I'm having the same result. It's not working. It was working fine with the test device that was installed during pilot phase (a Sagemcom T211). I've contacted Groupe E and they confirmed there is a bug in current firmware. It should be fixed later this fall. Here is a table I received explaining the problem:

[image]

So I'm afraid it will be necessary to find a way to have something between the Sagemcom device and HA to fix the errors.

Leefex, 2024-04-25:
Personally, I use the HomeWizard Energy P1 Wi-Fi adapter. From HA, I access the values via an URL, so I can retrieve all the meter values. This adapter costs around 28 euros. The link below gives more details:

https://www.objetsconnectes.be/2021/02/26/jeedom-sagemcom-ores-p1/

jacoch, 2024-04-25:
That's an interesting option that seems easy to use. On my side, as I'm already using MQTT for some of my sensors, I wanted to try that alternative : [https://github.com/antonijn/mqtt4dsmr](url).

jacoch, 2024-04-25:
Sure. Would be a better option. But Groupe E here in Switzerland is supposed to use Belgium version. And after comparing data sent by XT211 and expected data, I've noticed that another line is wrong. It's that one:

0-0:0.0.0(123456789012)

It's supposed to start with 0-0:96.1.1. After modifying the parser to read equipment id from that value and ignore the missing data in 0-0:98.1.0, everything works again. It's not perfect as the parser is in a docker that need to be fixed on each restart, but at least, I have a solution until Sagemcom fixes XT211.

binksjj, 2024-04-25:
I don't have the skills to understand the problem. But is it not possible to add the missing OBIS codes to the esphome code?

@jacoch , Could you explain the solution you implemented? So what is this parsers.py file?

jacoch, 2024-04-26:
Sure. Unfortuntately, I'm not using ESPhome for my P1 meter. I'm connected using USB. So the data is parsed directly by Home Assistant DSMR integration. But in the back, most solutions use a python library name [dsmr_parser](https://github.com/ndokter/dsmr_parser/tree/master/dsmr_parser). In my case, it's in the Synology docker. So I've edited the required files in that library to make Home Assistant work again. If ESPhome uses the same library, I can post the details about the required modifications.

binksjj, 2024-05-20:
> The whole DSMR component should be refactored so that any custom OBIS code could be configured.

I created an external component based on the standard [DSMR component](https://github.com/esphome/esphome/tree/dev/esphome/components/dsmr) from ESPHome and the [arduino-dsmr](https://github.com/glmnet/arduino-dsmr) (a forked from https://github.com/matthijskooijman/arduino-dsmr) from Guillermo Ruffino  (@glmnet) used by ESPHome.

Now, I can read all the datas from my SAGEMCOM XT211 smart meter installed by Groupe-E directly in Home Assistant  with esphome and the SlimmeLezer+ module, until the smart meter firmware update.

For those interested, here are my files: [my_components.zip](https://github.com/esphome/feature-requests/files/15381805/my_components.zip)

toggm, 2024-06-06:
@binksjj Just integrated your custom component. Works like a charm, many thanks.
If others are interested: Beside adding the extracting the component to the `config` directory you need to add the custom component location to the esp config:
```
external_components:
  - source:
      type: local
      path: my_components
```

And then add the group-e specific sensors to your configuration:
```
    energy_delivered_tariff1_groupe_e:
      name: "$human_devicename - Energy Consumed Tariff 1 Group-e"
      state_class: total_increasing
    energy_delivered_tariff2_groupe_e:
      name: "$human_devicename - Energy Consumed Tariff 2 Group-e"
      state_class: total_increasing
    energy_returned_tariff1_groupe_e:
      name: "$human_devicename - Energy Produced Tariff 1 Group-e"
      state_class: total_increasing
    energy_returned_tariff2_groupe_e:
      name: "$human_devicename - Energy Produced Tariff 2 Group-e"
      state_class: total_increasing
```

martignoni, 2024-09-21:
Up-to-date code in upstream DMSR component was merged to support OBIS codes used in Switzerland by Groupe E. Any chance this will be integrated?

malikbrh, 2024-10-31:
> Hello @razowski,
>
> To activate the P1 port, I sent several emails to customer service, made several calls over several months and I even wrote a registered letter to the management of the groupe-e. [...]

Hi @binksjj,
Do you have the complete Activation form somewhere? I would like to see the details they talk about on page 2 but it isn't in the PDF file you sent.
> Attention: La version actuelle de notre firmware actuel du compteur n’étant pas encore
parfaitement conforme au standard P1 DSMR (plus de détails en page 2).

Thanks in advance !

martignoni, 2024-10-31:
Here's the complete version of the form, with page 2 and a few details.

[Demande d'activation Port 1.pdf](https://github.com/user-attachments/files/17589988/Demande.d.activation.Port.1.pdf)

malikbrh, 2025-06-05:
@nwaelti Same here ! Firmware upgrade rolled out less than a month ago, they are now DSMR compliant. But thanks for the tip !

martignoni, 2025-06-05:
@nwaelti Same for me, just like @toggm and @malikbrh.