esphome/feature-requests #2570: Arduino ESP32 nano support

maccampb, 2024-01-21:
**Describe the problem you have/What new integration you would like**

Please add support in ESPHome for the new Arduino ESP32 Nano.

**Please describe your use case for this integration and alternatives you've tried:**

This is a new board in the esp32 community and should be supported for the additional capabilities.
**Additional context**

https://docs.arduino.cc/resources/datasheets/ABX00083-datasheet.pdf

tsmith512, 2024-02-01:
Also interested in this --- per https://docs.platformio.org/en/latest/boards/espressif32/arduino_nano_esp32.html PlatformIO does support it but it was added in platformio/platform-espressif32#1162 in v6.4.0. Looks like esphome may be on v6.1.

Skywalker8510, 2024-08-30:
> Looks like esphome may be on v6.1.

actually they might be on v5.4 based off of this code.

```# The platformio/espressif32 version to use for arduino frameworks
#  - https://github.com/platformio/platform-espressif32/releases
#  - https://api.registry.platformio.org/v3/packages/platformio/platform/espressif32
ARDUINO_PLATFORM_VERSION = cv.Version(5, 4, 0)```