esphome/feature-requests #2573: autoclose value for time_based cover

djsomi, 2024-01-23:
An autoclose value for time_based cover would be nice.

Most of the Gate electronics works this way: After fully opened there is a time to wait  and gate closing automatically. 
If using RF signals only to operate, and does not have any opened/closed binary state input, an autoclose time would be useful to get the gate status back to Closed after the defined time. 

Maybe it can be coded within the actual framework, if yes please share a working solution.

nagyrobi, 2024-01-23:
```
cover:
  - platform: ...
    id: my_gate
    # ...
    on_open:
      - logger.log: "Gate Open!"
      - delay: 3min #you can use a lambda here, pointing at a number if you want it configurable
      - logger.log: "Gate Closing!"
      - cover.close: my_gate
```

nagyrobi, 2024-01-24:
I suggest to ask on Discort support channel. There create a thread, and paste your entire config to see exactly what can be done.

closed 2024-01-24 by nagyrobi