esphome/feature-requests #2578: Add docs about ST7789v2 display

iBobik, 2024-01-25:
**Describe the problem you have/What new integration you would like**

I do not know how to configure platform and pins.

**Please describe your use case for this integration and alternatives you've tried:**

I want to use M5Stack Station's display, but it is not clear how to configure it:
- ST7789v2 is not in supported list, but maybe M5STACK or ST7789V is compatible
- `dc_pin` is required to be configured, but there is no such pin in Station's docs

**Additional context**

https://docs.m5stack.com/en/core/station_485
[image]

[image]

tiaanv, 2024-08-27:
Did you get any feedback on this?  Is it supported?

nvtkaszpir, 2025-07-03:
yes, see https://github.com/esphome/issues/issues/3591#issuecomment-2569732913

closed 2025-07-07 by clydebarrow