esphome/feature-requests #2583: Support for AXP2101 Single Cell NVDC PMU with E-gauge

lboue, 2024-01-29:
**Describe the problem you have/What new integration you would like**

AXP2101 Single Cell NVDC PMU with E-gauge
**Please describe your use case for this integration and alternatives you've tried:**

M5Stack Core2 v1.1
M5Stack CoreS3

**Additional context**

 * [M5Unified AXP2101 class](https://github.com/m5stack/M5Unified/blob/master/src/utility/AXP2101_Class.cpp
)

lboue, 2024-01-29:
I found this ESPHome componant: [esphome-axp2101 ](https://github.com/stefanthoss/esphome-axp2101)

BigBobbas, 2024-04-17:
I would also like support for the AXP2101 for Lilygo T-watch S3 using esp-idf framework

titchy, 2025-03-27:
I also have a need for this for the Lilygo T-Camera S3. There is example code that uses https://github.com/lewisxhe/XPowersLib.git
It also requires the now no longer supported custom component feature and a header file here: https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/blob/master/esphome/AXP2101_component.h
This includes voltages required by the camera board etc.

monsieurlatte, 2025-04-30:
Would love to see this as well for my m5stack core 2 v1.1