esphome/feature-requests #2585: Support vga display with bitluni/ESP32Lib

frateche, 2024-01-30:
**Describe the problem you have/What new integration you would like**

Integration of bitluni/ESP32Lib vga driver in esphome.

**Please describe your use case for this integration and alternatives you've tried:**

It could be nice to use a basic vga screen as display component. 3bits mode is very easy to wire.

**Additional context**

https://github.com/bitluni/ESP32Lib
https://www.youtube.com/watch?v=qJ68fRff5_k

docteurzoidberg, 2024-05-26:
more globally, esphome is missing a "tvout" / composite library for retrofitting purposes. i would tend to prefer fabgl's over bitluni's one for board availability + docs

plus fabgl supports lvgl which will be soon to be released.

mrivero33, 2024-08-23:
Yes, please!

cosmicDustOfLightLength, 2025-03-10:
https://github.com/bitluni/ESP32-S3-VGA