esphome/feature-requests #2586: Reduce power consumption ESP8266

Uhwah, 2024-01-31:
**Describe the problem you have/What new integration you would like**
 I did various tests with ESPHome and Tasmota. I noticed that ESPHome doesn't seem to have any power saving functions (except Deep Sleep). Tasmota has a feature called Dynamic Sleep. My measurements have shown that an ESP8266 with a DS18b20 sensor under ESPHome has 2.4 times the power consumption compared to Tasmota. The whole thing also causes the ESP8266 to become noticeably warm under ESPHome, which in my opinion can influence temperature measurements.

**Please describe your use case for this integration and alternatives you've tried:**
 Reducing power consumption should always be a high priority these days.

**Additional context**

twasilczyk, 2024-01-31:
This is most likely not a problem with power saving functions, but floating GPIO power draw. PTAL https://github.com/esphome/feature-requests/issues/2587

Uhwah, 2024-02-01:
I changed all GPIO pins as described.
(GPIO 16 was prevented with an error message)-> GPIO Pin 16 does not support pullup pin mode. Please choose another pin.
Unfortunately nothing has changed.  
The ESP01s consistently requires 0.05A even though the sensor query takes 60s (averaged 0,255W).
With Tasmota I can see how the consumption fluctuates with a maximum of 0.05A (averaged 0,106W).

- My yaml code:

esphome:
  name: esp-01
  friendly_name: ESP-01

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: xxx

ota:
  password: xxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-01 Fallback Hotspot"
    password: xxx

captive_portal:

binary_sensor:
  - platform: gpio
    id: gpio02
    pin:
      number: GPIO02
      mode:
        input: true
        pullup: true
  - platform: gpio
    id: gpio14
    pin:
      number: GPIO14
      mode:
        input: true
        pullup: true
  - platform: gpio
    id: gpio12
    pin:
      number: GPIO12
      mode:
        input: true
        pullup: true
  - platform: gpio
    id: gpio13
    pin:
      number: GPIO13
      mode:
        input: true
        pullup: true
  - platform: gpio
    id: gpio03
    pin:
      number: GPIO03
      mode:
        input: true
        pullup: true
  - platform: gpio
    id: gpio05
    pin:
      number: GPIO05
      mode:
        input: true
        pullup: true
  - platform: gpio
    id: gpio04
    pin:
      number: GPIO04
      mode:
        input: true
        pullup: true
  - platform: gpio
    id: gpio00
    pin:
      number: GPIO00
      mode:
        input: true
        pullup: true
  - platform: gpio
    id: gpio15
    pin:
      number: GPIO15
      mode:
        input: true
        pullup: true

    
dallas:
  - pin: GPIO1

sensor:
  - platform: dallas
    address: 0xaf00000dff1b6728
    name: "Vorlauftemperatur"
  - platform: dallas
    address: 0x8d00000dfeb1d828
    name: "Rücklauftemperatur"
  - platform: dallas
    address: 0x630000007ca34a28
    name: "Warmwasserspeicher Temperatur"

twasilczyk, 2024-02-01:
GPIO16 supports pull-down. But if it doesn't help, then it might be a different problem.

bkbartk, 2024-02-22:
> The ESP01s consistently requires 0.05A even though the sensor query takes 60s (averaged 0,255W).
> With Tasmota I can see how the consumption fluctuates with a maximum of 0.05A (averaged 0,106W).

so there is this option in esphome
https://esphome.io/components/wifi.html#wifi-power-save-mode
which i do not see in your script, but I know tasmota has a different default here and therefor by default consumes less energy

Uhwah, 2024-02-25:
That was the right hint! Now the value is very close to the value of Tasmota. Many thanks for the help.

fightforlife, 2025-07-25:
See this issue: https://github.com/esphome/issues/issues/1532
It seems High and Low are twisted in ESP8266. Might be good to try both.