esphome/feature-requests #2587: Initialize all unused GPIO with pull-ups to prevent power draw

twasilczyk, 2024-01-31:
**Describe the problem you have**
Floating GPIO causes unnecessary power draw and instability

**Please describe your use case for this integration and alternatives you've tried:**
Alternative: manually setting all unused GPIO to input with pull-up

**Additional context**
https://github.com/esphome/issues/issues/1950
https://github.com/esphome/issues/issues/5085
https://github.com/esphome/feature-requests/issues/2586
https://www.reddit.com/r/Esphome/comments/ppzd08/is_it_normal_for_a_smart_plug_to_feel_warmer_when/
https://community.home-assistant.io/t/esp-overheat-without-ha/508317/3
https://community.home-assistant.io/t/device-overheating-on-esphome-but-fine-on-tasmota/464181/8

**Suggested feature**
ESPHome should enumerate all unused GPIOs (I think it should have that information in board config?) and set default for all as GPIO input with pull-up. Example from https://github.com/esphome/issues/issues/5085#issuecomment-1897557212:
```
binary_sensor:
  - platform: gpio
    id: gpio16
    pin:
      number: GPIO16
      mode:
        input: true
        pullup: true
```

twasilczyk, 2024-02-08:
After measurements (measuring Gosund WP5 plug with Sonoff S31 power-metering plug) I can say that setting up pull-ups/downs saves about 0.45W in the case of my plug.

nagyrobi, 2024-02-09:
That's a lot!

kbx81, 2024-02-09:
0.45W doesn't make any sense. An ESP32 would only use maybe 0.65W or so at its _peak._ None of my ESP32 boards draw 0.45W typically -- only 0.2 - 0.3 watts, with small spikes when the radio transmits.

CarlosGS, 2024-02-09:
@twasilczyk In one comment you mention that power_save_mode doesn't make a measurable difference, could you double check please? I did some tests a while ago that should still apply: https://github.com/esphome/issues/issues/1532#issuecomment-718669014

Also, can you confirm that this only affects ESP8266/ESP8285, or have you also observed this in any ESP32 board?

twasilczyk, 2024-02-09:
It's possible that my measurements are a bit off - according to a datasheet, CSE7766 (Sonoff S31 power metering chip) absolute error for measurements under 15W is 0.5W. But I did my best (measured with a constant load of a small light bulb) and the power consumption before and after the fix is very noticeable on HA diagrams, while Wi-Fi power saving gain is below CSE7766 readings noise. Right now I have a build with default Wi-Fi settings (no power saving at all) and the plug is at the room temperature.

This is very much aligned with [my observations](https://github.com/esphome/issues/issues/1950#issuecomment-1919827544) on plug body temperature. When I looked inside (of course, after un-plugging the plug from 110V), not only ESP8266 was warm but also the power supply section (condensators and inductor).

I didn't try any ESP32 boards - only a single ESP8266 off-the-shelf product.

twasilczyk, 2024-02-12:
`Pull-up/dn` is with pull ups/downs added and `Pull-up + Wi-Fi` is pull ups/downs *and* Wi-Fi power saving settings. IMO the difference is negligible compared to the jump between before-fix and pull ups/downs. I'm changing all unused GPIOs, but I had similar results only changing `GPIO15` (not 16!).

Please see *minimal* code examples below. Unfortunately I can't make them really minimal - I don't want to take the recovery mode config out, because flashing these manually is extremely difficult.
[v1-before-fix.yaml.txt](https://github.com/esphome/feature-requests/files/14251500/v1-before-fix.yaml.txt)
[v2-pull-up-dn.yaml.txt](https://github.com/esphome/feature-requests/files/14251501/v2-pull-up-dn.yaml.txt)
[v3-pull-up-wifi.yaml.txt](https://github.com/esphome/feature-requests/files/14251502/v3-pull-up-wifi.yaml.txt)