esphome/feature-requests #2589: 4th fixed position beside open, stop, close in Template Cover.

Stipena, 2024-02-02:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2024-02-03:
Can you show us what kind of covers are these? How do they look like? Seems from your description like not everywhere in the world can be found.

Stipena, 2024-02-03:
> Can you show us what kind of covers are these? How do they look like? Seems from your description like not everywhere in the world can be found.

Sure. While moving up or down, non-transparent textile lines intersect.

https://github.com/esphome/feature-requests/assets/4146267/40768821-befc-421e-8383-e6af0f3f7e67

nagyrobi, 2024-02-03:
What's the difference between middle and stop in your thinking?

Stipena, 2024-02-03:
Stop is used for manual usage, like example if you want to open/close blinds to 1/3. And of course like emergency stop if endstop failed for some reason (had one in other project).
Middle position is automated, you just press once without looking at it and you know that it will stop where you wanted it to stop. It will be pre-calibrated position.

ssieb, 2024-02-04:
Just make a button that sets the cover position to that point.

Stipena, 2024-02-23:
> Just make a button that sets the cover position to that point.

Can I make it within covers template? Or I will need to make that button in separate card in lovelace? If in separate card then that's not what I'm asking.