esphome/feature-requests #2590: Support for HLW8112

kostasdimos, 2024-02-03:
**Describe the problem you have/What new integration you would like**
I have purchased a power sensor with two clamps. The model on the box is described as PJ-1103 A.
On my search it uses the cb2s as wifi module and the hlw8112 power sensor chip.
I have a plan to replace the cb2s with esp chip and to flash with esp home.
But as I can see there is not yaml configuration for the hlw8112 chip.

poldim, 2024-03-06:
I'm in the same boat. My device is labeled PJ1203A V8 on the PCB. 
There are only 5 pins attaching the CB2S to the main PCB. So it should be easy to swap in an ESP.

[image]
[image]
[image]

mpissarello, 2024-05-05:
Hi all!
Same here with my power meter Earu PC311-W-TY.

The WiFi chip is a WBR3 (W701-VA2-CG):
https://developer.tuya.com/en/docs/iot/wbr3-module-datasheet?id=K9dujs2k5nriy
FWIU this chip is not supported ATM, but it's swappable with an ESP-12S (of F)

The energy measure chip is HLW8112, and it's connected via SPI:
https://redmine.laas.fr/attachments/download/3444/DS_HLW8110_HLW8112_EN_Rev1.01%20.pdf

I found this Git with a preliminary driver for the HLW8112, may be ported to ESPHome?
https://github.com/MahdaSystem/HLW811x

[image]
[image]
[image]
[image]

nschimme, 2024-09-12:
@Hossein-M98 has created a driver here: https://github.com/MahdaSystem/HLW811x

Hossein-M98, 2024-09-13:
> @Hossein-M98 has created a driver here: https://github.com/MahdaSystem/HLW811x

I am so sorry. The driver has not been completed. I am working on other projects and do not have time to complete this driver. The HLW811x has bad documentation and took several days from me. So we decided to drop it for a while and work on other important projects.

> @Hossein-M98 would you have any interest integrating your driver into the esphome ecosystem?

I do not have any idea about esphome :)

artemdavydenko, 2024-09-24:
> Here another generic Aliexpress Tuya Meter that has the HLW8112.
> 
> PCB version is PM01_A002 and datecode is 20240705.
> 
> The WiFi Chip is Tuya - CBU which is supported in ESPHome.
> 
> ![IMG_20240827_044834_253](https://private-user-images.githubusercontent.com/48742980/361516654-ff08b5de-4f84-4fb7-9252-a2ffeffaec58.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxNjk2NjIsIm5iZiI6MTcyNzE2OTM2MiwicGF0aCI6Ii80ODc0Mjk4MC8zNjE1MTY2NTQtZmYwOGI1ZGUtNGY4NC00ZmI3LTkyNTItYTJmZmVmZmFlYzU4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI0VDA5MTYwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRlM2Q4OGE0ZDRiY2I4MGQ1NWMxMmE5Y2E4ZGFjYjg0NmI3ZTBmYmQwM2YwYjU1MDZkZTY2YTE5ZDQyNzI1NjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.RxDpsmSK0EuQn0DlBgbHirhmsE5diUv5ScNLlcSL-qs) ![20240827_045423](https://private-user-images.githubusercontent.com/48742980/361516644-aba74845-2531-4db6-92b6-4298299dcea8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxNjk2NjIsIm5iZiI6MTcyNzE2OTM2MiwicGF0aCI6Ii80ODc0Mjk4MC8zNjE1MTY2NDQtYWJhNzQ4NDUtMjUzMS00ZGI2LTkyYjYtNDI5ODI5OWRjZWE4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI0VDA5MTYwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFmZDY5MjI4YWE2NmI0OTkwNTExMTdkM2IwOTkwZDYyNjc5ZTcxMzQ0ZmVkYjA5NDQ4MGQ0YWJkNWY4ZDYzMTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.xlUgFwBHiq9pNYwtukz0aG-vdh4BC-pEuAG63ap6UQc) ![IMG_20240827_044929_495](https://private-user-images.githubusercontent.com/48742980/361516659-b9ba2635-efe5-4d14-9c95-d142c0bd3b77.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxNjk2NjIsIm5iZiI6MTcyNzE2OTM2MiwicGF0aCI6Ii80ODc0Mjk4MC8zNjE1MTY2NTktYjliYTI2MzUtZWZlNS00ZDE0LTljOTUtZDE0MmMwYmQzYjc3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI0VDA5MTYwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRmZWE1MGFkZmQ3N2ZhZWI5NTZkN2NkZGFhMTdiMjlkNTVmNjg3YmQ1ZTU1M2M1ZmZiNzhiYzMyYjMwZGNjMDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.AAFU2Y68BQjum91VyDhxbd10DD2AdSIQQqqIf1lPcNs) ![20240827_051137](https://private-user-images.githubusercontent.com/48742980/361516650-bd501339-f3fc-4c2b-8df4-e63b9a10bcd9.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxNjk2NjIsIm5iZiI6MTcyNzE2OTM2MiwicGF0aCI6Ii80ODc0Mjk4MC8zNjE1MTY2NTAtYmQ1MDEzMzktZjNmYy00YzJiLThkZjQtZTYzYjlhMTBiY2Q5LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI0VDA5MTYwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU0YWE4NGRkZjc2MTdmZmYyNmI4YjkzNDdlMGNjYjRlZGZkMjM1NmJlYjU2OTU3NDI4YzA5ZmU3YmMxNjc5MWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Rz3q5w14qO8XIzPQ1D5hi-p1MDCYk0LyfBzZXFsA0QA)
> 
> I had some spare time and traced the connections between the CBU and HLW8112: ![image](https://private-user-images.githubusercontent.com/48742980/361532707-00107beb-8dc4-467b-af51-84bc38b4117e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxNjk2NjIsIm5iZiI6MTcyNzE2OTM2MiwicGF0aCI6Ii80ODc0Mjk4MC8zNjE1MzI3MDctMDAxMDdiZWItOGRjNC00NjdiLWFmNTEtODRiYzM4YjQxMTdlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI0VDA5MTYwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkxYmQ4YTQ5YWQ3Mzk4NmZhMTc5NGQ2MGQwYmQ5MWFiODdkMzg3Nzc3YzQyY2Y2NGQ2NmE3ODM4NDhhYzkyM2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.GDwPaKW3TSszPJcASyd1j4rECNm5Mosipb3K-j994H0)

Do you have the possibility to make a full dump of this device? I have the same device, but it's been flashed without the original firmware, and I'm now trying to find it. Thanks in advance.

CladZo91, 2024-09-24:
I think you're in luck, while i flashed it as well, I dumped the original firmware with ltchiptool.

Vortecsz, 2025-01-02:
Any update guys?
Did someone managed to use it with esphome?

Black6spdZ, 2025-01-12:
> Here another generic Aliexpress Tuya Meter that has the HLW8112.
> 
> PCB version is PM01_A002 and datecode is 20240705.
> 
> The WiFi Chip is Tuya - CBU which is supported in ESPHome.
> 
https://www.elektroda.com/rtvforum/topic4016609-30.html

mpissarello, 2025-01-12:
> https://www.elektroda.com/rtvforum/topic4016609-30.html

[Comment #26](https://www.elektroda.com/rtvforum/viewtopic.php?p=21333858#21333858) of this topic mentions an HLW811x driver, written for the BabyOS firmware:

[https://github.com/notrynohigh/BabyOS/blob/master/bos/drivers/b_drv_hlw811x.c](https://github.com/notrynohigh/BabyOS/blob/master/bos/drivers/b_drv_hlw811x.c)

Perhaps this can be ported to ESPHome, instead of the Hossein-M98 one?

fcol95, 2025-03-21:
Also curious about updates on this!
Got the PC311-W-TY energy meter with WBR3 chip and HLW8112 too.
Very green on all esphome codebase stuff and unsure if functional or status of the work, but seems this guy started integrating the driver in OpenBeken if that can help
https://github.com/dicamarques14/OpenBK7231T_App/tree/drv_hlw8112.
I did a quick and dirty diagram of the WBR3 connection on that energy meter if that can help too!
[image]

kboran, 2025-10-02:
There is yet another driver written as part of an EV charging project. I haven't taken a close look yet but it might be helpful. https://github.com/pazzk-labs/evse/blob/main/src/metering/adapter/hlw8112.c

BmdOnline, 2025-11-10:
I have [this version too](https://github.com/esphome/feature-requests/issues/2590#issuecomment-2310689330).

According to this diagram and HLW8112 pinout, PIN 12 is factory set to ON.
[image]
[image]
[image]
Which means that **SPI mode must be used, and not UART mode**.

ESPHome doesn't support SPI for libretiny platform (BK7231 devices).
> This feature is only available on [<Platform.ESP32: 'esp32'>, <Platform.ESP8266: 'esp8266'>, <Platform.RP2040: 'rp2040'>].

According to [this thread](https://www.elektroda.com/rtvforum/topic3945667-30.html#20478523), it seem that 
> The beken_os SPI drivers did not work. I think the programmers started with the drivers but didn't finish them.

I've tried to explore [@Hossein-M98](https://github.com/Hossein-M98) [HLW811x Library](https://github.com/MahdaSystem/HLW811x).
I'm stuck rewriting `HLW811x_platform` because of SPI driver.
I need to find a working Beken SPI driver that can be ported to HLW811x Library `HLW811x_platform`.

BmdOnline, 2026-08-30:
Hi,
I've published an update, working with original [CBU  version](https://github.com/BmdOnline/esphome_hlw811x).
I'm using bit-banging method for SPI on Beken BK7231N, so everything works now.

tablatronix, 2026-11-06:
What about the versions like the OPs, this would not be SPI would have to use uart, what was the solution for the PJ1203A/PJ1103C.
I flashed the  T1-M not sure what I can do now...

tablatronix, 2026-11-06:
Ok I got it flashed and logging

```php
[21:21:15.558][C][mdns:324]:   Hostname: pj1103c
[21:21:15.579][C][libretiny:015]: LibreTiny:
[21:21:15.579][C][libretiny:015]:   Version: v1.13.0+sha.6514b26 on t1-m, compiled at Nov  6 2026 19:54:07, GCC 10.3.1 (-O1)
[21:21:15.579][C][libretiny:015]:   Loglevel: 3
[21:21:15.582][C][libretiny:021]:   Optimization: -Os, SDK: -O1
[21:21:15.602][C][tuya:062]: Tuya:
[21:21:15.606][C][tuya:079]:   Datapoint 132: switch (value: OFF)
[21:21:15.606][C][tuya:081]:   Datapoint 116: int value (value: 1000)
[21:21:15.609][C][tuya:081]:   Datapoint 122: int value (value: 1000)
[21:21:15.614][C][tuya:081]:   Datapoint 129: int value (value: 10)
[21:21:15.622][C][tuya:079]:   Datapoint 133: switch (value: OFF)
[21:21:15.627][C][tuya:079]:   Datapoint 134: switch (value: OFF)
[21:21:15.648][C][tuya:081]:   Datapoint 137: int value (value: 2600)
[21:21:15.648][C][tuya:081]:   Datapoint 138: int value (value: 850)
[21:21:15.649][C][tuya:081]:   Datapoint 123: int value (value: 1000)
[21:21:15.649][C][tuya:081]:   Datapoint 124: int value (value: 1000)
[21:21:15.669][C][tuya:081]:   Datapoint 125: int value (value: 1000)
[21:21:15.669][C][tuya:081]:   Datapoint 128: int value (value: 1000)
[21:21:15.669][C][tuya:079]:   Datapoint 149: switch (value: OFF)
[21:21:15.669][C][tuya:079]:   Datapoint 150: switch (value: OFF)
[21:21:15.675][C][tuya:081]:   Datapoint 145: int value (value: 150000)
[21:21:15.683][C][tuya:081]:   Datapoint 146: int value (value: 330000)
[21:21:15.687][C][tuya:081]:   Datapoint 117: int value (value: 1000)
[21:21:15.692][C][tuya:081]:   Datapoint 118: int value (value: 1000)
[21:21:15.698][C][tuya:081]:   Datapoint 119: int value (value: 1000)
[21:21:15.704][C][tuya:081]:   Datapoint 127: int value (value: 1000)
[21:21:15.710][C][tuya:079]:   Datapoint 135: switch (value: OFF)
[21:21:15.715][C][tuya:079]:   Datapoint 136: switch (value: OFF)
[21:21:15.724][C][tuya:081]:   Datapoint 139: int value (value: 150000)
[21:21:15.735][C][tuya:081]:   Datapoint 140: int value (value: 330000)
[21:21:15.735][C][tuya:081]:   Datapoint 130: int value (value: 0)
[21:21:15.740][C][tuya:081]:   Datapoint 108: int value (value: 0)
[21:21:15.744][C][tuya:081]:   Datapoint 106: int value (value: 0)
[21:21:15.756][C][tuya:081]:   Datapoint 131: int value (value: 0)
[21:21:15.760][C][tuya:081]:   Datapoint 109: int value (value: 0)
[21:21:15.760][C][tuya:081]:   Datapoint 107: int value (value: 0)
[21:21:15.768][C][tuya:081]:   Datapoint 112: int value (value: 1223)
[21:21:15.778][C][tuya:081]:   Datapoint 111: int value (value: 5875)
[21:21:15.785][C][tuya:081]:   Datapoint 114: int value (value: 0)
[21:21:15.785][C][tuya:081]:   Datapoint 105: int value (value: 0)
[21:21:15.789][C][tuya:081]:   Datapoint 121: int value (value: 100)
[21:21:15.794][C][tuya:081]:   Datapoint 113: int value (value: 0)
[21:21:15.800][C][tuya:081]:   Datapoint 101: int value (value: 0)
[21:21:15.805][C][tuya:081]:   Datapoint 110: int value (value: 100)
[21:21:15.811][C][tuya:081]:   Datapoint 115: int value (value: 0)
[21:21:15.817][C][tuya:097]:   Product: '{"p":"wifech3utowiyknu","v":"6.1.2","m":2}'
[21:21:15.830][D][tuya:407]: Datapoint 132 update to OFF
```

tablatronix, 2026-11-06:
Found an autoexec

https://www.elektroda.com/rtvforum/topic4016609-60.html#21781239

Not sure how to turn this into esphome config but will match em up

tablatronix, 2026-11-06:
Ok might have found it, 
https://www.reddit.com/r/homeassistant/comments/1qqkrv0/how_to_configure_the_pj1103_dual_phase_power_meter/

kostasdimos, 2026-11-06:
As i can see there are two versions of the board. The one has tuya cbu chip and the tuya cb2s. both of them cam be replaced with the ESP32-C3 and the ESP-02S.
i have already replaced the chips on two devices with cb2s successfully and ordered another two. unfortunately the two new came with the cbu chip.  while trying to remove the cbu with the soldering iron i destroyed both of them by cutting the lines on the pcb. i was so gentle with them but i dont know how it done.

My question is if you know where any online shop to buy the two meters that i need  witch have the cb2s chip that is easy for me to replace?

tablatronix, 2026-11-07:
Got it running now esphome sensors yaml, Ill post the whole esphome template somewhere useful

```yaml
    - platform: "tuya"
      name: power_factor_a
      device_class: power_factor
      accuracy_decimals: 2
      filters:
        multiply: 0.01
      sensor_datapoint: '110'

    - platform: "tuya"
      name: power_a
      unit_of_measurement: W
      device_class: power
      accuracy_decimals: 2
      filters:
        multiply: 0.1
      sensor_datapoint: '101'

    - platform: "tuya"
      name: power_b
      unit_of_measurement: W
      device_class: power
      accuracy_decimals: 2
      filters:
        multiply: 0.1
      sensor_datapoint: '105'

    - platform: "tuya"
      name: energy_forward_a
      unit_of_measurement: KWh
      device_class: power
      accuracy_decimals: 2
      filters:
        multiply: 0.01
      sensor_datapoint: '106'

    - platform: "tuya"
      name: energy_forward_b
      unit_of_measurement: KWh
      device_class: power
      accuracy_decimals: 2
      filters:
        multiply: 0.01
      sensor_datapoint: '108'

    - platform: "tuya"
      name: freq
      unit_of_measurement: Hz
      device_class: frequency
      accuracy_decimals: 2
      filters:
        multiply: 0.01
      sensor_datapoint: '111'

    - platform: "tuya"
      name: voltage_a
      unit_of_measurement: V
      device_class: voltage
      accuracy_decimals: 2
      filters:
        multiply: 0.1
      sensor_datapoint: '112'

    - platform: "tuya"
      name: power_factor_b
      device_class: power_factor
      accuracy_decimals: 2
      filters:
        multiply: 0.01
      sensor_datapoint: '121'

    - platform: "tuya"
      name: forward_energy_total
      unit_of_measurement: KWh
      device_class: energy
      accuracy_decimals: 2
      filters:
        multiply: 0.01
      sensor_datapoint: '130'

    - platform: "tuya"
      name: total_power
      unit_of_measurement: W
      device_class: power
      accuracy_decimals: 2
      filters:
        multiply: 0.1
      sensor_datapoint: '115'

    - platform: "tuya"
      name: current_a
      unit_of_measurement: A
      device_class: current
      accuracy_decimals: 2
      filters:
        multiply: 0.001
      sensor_datapoint: '113'

    - platform: "tuya"
      name: current_b
      unit_of_measurement: A
      device_class: current
      accuracy_decimals: 2
      filters:
        multiply: 0.001
      sensor_datapoint: '114'

    - platform: "tuya"
      name: report_rate_control
      unit_of_measurement: s
      sensor_datapoint: '129'
```