esphome/feature-requests #2591: Configurable improv_serial firmware name and version identifier

DanielBaulig, 2024-02-04:
**Describe the problem you have/What new integration you would like**

When building a firmware image using ESPHome and the improv_serial component, ESPHome will configure the improv version response to be "ESPHome", the ESPHome version used to build the firmware and the device name. The code for this -- if I'm not mistaken -- can be found here:
https://github.com/esphome/esphome/blob/dev/esphome/components/improv_serial/improv_serial_component.cpp#L166-L170
I would like both the version string and firmware identifier to be configureable from YAML. 

**Please describe your use case for this integration and alternatives you've tried:**

When using improv and ESP Web Tools to flash the resulting firmware image to a MCU, these strings are presented to the user to identify the firmware and firmware version. This makes sense for a lot of use-cases, but in others one might want to use the use-case specific firmware's name and version. For example, I am using ESPHome to create a firmware image for a device I am distributing to end-users. I would like the firmware to identify itself via improv as the use-case specific firmware and use my own versioning system.

**Additional context**

Currently however, this is not possible. The improv_serial component does not allow for these values to be configured or overridden from the ESPHome yaml.

I would see two possible options to implement an override mechanism:

1) An explicit override configuration through the improv_serial component. E.g. through optional `firmware_name` and `firmware_version` configuration fields. If not provided, the ESPHome defaults are used, otherwise the values provided through the configuration fields are used.

2) Alternatively `project: name` and `project: version` from the `esphome` could be used. An additional configuration field (e.g. `identify_with_project`) in improv_serial might be introduced to enable this override, which should be off by default. So for existing configurations they would continue using the ESPHome defaults, even if they have project_name and project_version currently set.

I was wondering if this is an oversight or just so far has not been a priority or if this is a specific conscious decision? Would a pull request to add the option to override the firmware identifier and version from YAML be something that the project is willing to accept? If so, some feedback on the proposed approach would be greatly appreciated.

DanielBaulig, 2024-02-09:
Any thoughts on this? I'd love to provide a PR but also don't want to spend too much time building it just to have it be shot down as something that the project won't take later.

AzonInc, 2024-08-11:
I came across the same topic when trying to provide ready made firmware. ESP Web Tools are not able to recognize it other than ESPHome without the projects version. I created my own improv_serial component to solve this but it would be way better to have it working out of the box.

closed 2024-08-12 by jesserockz