esphome/feature-requests #2594: Request for PWM Direct Drive speaker support.

whiteknight6766, 2024-02-05:
**Describe the problem you have/What new integration you would like**

I am attempting to setup a "hello world" for a variant of esp32-c3-lcdkit, found at link 1. I have been able to get everything working except the audio. ESP Home is requiring a i2c platform for the speaker: integration. However the NS4150 chipset, Link2, is a low powered mono direct drive pwm speaker. Can we add a second class of amplifier to the audio section? 

Link1 :https://www.amazon.com/dp/B0C9M1887C?psc=1&ref=ppx_yo2ov_dt_b_product_details
Link2 https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/_static/esp32-c3-lcdkit/datasheets/NS4150B.pdf
**Please describe your use case for this integration and alternatives you've tried:**

Use case will be to support low powered speakers, that do not otherwise have a digital control chip 

**Additional context**

Links 1 and 2 above are my hardware, but I am requesting a general purpose PWM Direct drive mode that would be a generic speaker driver. even allowing the speaker section to optionally depend on a generic pwm class other than just i2c.

ssieb, 2024-02-06:
I think you mean `i2s`, not `i2c`.  As far as I can tell, that board uses i2s for the audio output.

whiteknight6766, 2024-02-07:
Yes i2s. and no this board is not i2s. from the reference material "The NS4150 has digital amplifiers with analog input" and multiple references to "mono direct drive PWM speaker". the block diagram for this board has only one connection via GPIO pin 3. I have confirmed this to be true with a multi-meter. only pin3 is connected to the esp32. Which means setting i2s_lrclk_pin: and i2s_bclk_pin: will do nothing. This was, however, my final answer. I assigned those two pins to NC and a test mp3 played. 

From my side, my feature stands. can we remove the i2c requirement for speaker: ? This would support low power PWM output to analog audio amplifiers. 

to anyone coming across this post after the inevitable "closed because a workaround was found" set the i2c to two different NC pins mine just happened to be 24,25 but there is nothing special about those pins.

ssieb, 2024-02-07:
But if you used esphome, then you still somehow used i2s (not i2c).  I don't understand how that worked, but that is also what I saw in the example code for that board.  What was the config you used?

DefenestratingWizard, 2025-03-01:
ESP_ADF seems to include code for a pwm'd audio output. Seeing as esphome seems to use esp-adf in other parts of the system (though its not listed as a component) it would be nice to have access to this to have the capability to play (very) low quality chimes/voices through buzzers instead of only the rtttl tones we currently have.

https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_play_mp3_with_dac_or_pwm/README.md

ssieb, 2025-03-01:
esp-adf is being removed.