esphome/feature-requests #2597: Add support for AS7343 spectral sensor

Beul85, 2024-02-06:
**Describe the problem you have/What new integration you would like**

Add support for the AS7343 spectral sensor.

**Please describe your use case for this integration and alternatives you've tried:**

Diy Par Meter Project.

**Additional context**

The older AS7341 is already integrated in esphome and nearly identical (11vs14 channels) to the AS7343.

HerzBeatz, 2024-02-06:
I would love to see this integrated into esphome.
With this sensor you can make a cheap PPFD device which costs $500 online .

latonita, 2024-02-25:
@Beul85 @HerzBeatz I can take care of it. Do you have esp32 and the sensor connected? 
I don't have the device, but with wireguard it will be easy for me to work on the driver.

Beul85, 2024-02-25:
Hello, 
esp32 and sensor as7343 are connected und wireguard is installed. I would love your help. what do i do next?

latonita, 2024-02-25:
Drop me an email Anton.viktorov at live.com

BR, Anton

Beul85, 2024-02-25:
I sendet you a email with reference esp32 as7343

latonita, 2024-02-25:
@Beul85 I made initial version, please check if it works at all
https://github.com/esphome/esphome/pull/6285

HerzBeatz, 2024-05-15:
@latonita is there any chance that you could do this PPFD calibation for the as7341 as well? it looks like that the PPFD reading from 7343 is very close to a high end PPFD reader.

latonita, 2024-05-20:
@HerzBeatz  did you try 7343 and compared to some calibrated devices?
Unfortunately its impossible to do proper measurements without 1) diffuser and 2) calculated glass attenuation factor.
Numbers out of the box can be very inacurate.
Plus sensor doesnt really work with high lux (1000+) without glass attenuation filters.
after i review and upload basic features I will look at 7341.

harpi4, 2026-08-18:
@latonita 

If this helps (otherwise, let me know what I can do):

Tested PR #18012 with an AS7343 on ESP32

```yaml
external_components:
  - source: github://pr#18012
    components: [as7341, as734x]
    refresh: 1h

defaults:
  sensor_name: "AS7343 Spectral"
  sensor_id: as7343_spectral
  sensor_i2c_id: bus_a
  sensor_address: 0x39

  sensor_gain: X4
  sensor_atime: 29
  sensor_astep: 599
  sensor_update_interval: 30s
  sensor_normalize_basic_counts: "ALL"
  sensor_glass_attenuation_factor: "1.0"

  sensor_f1_internal: false
  sensor_f2_internal: false
  sensor_fz_internal: false
  sensor_f3_internal: false
  sensor_f4_internal: false
  sensor_fy_internal: false
  sensor_f5_internal: false
  sensor_fxl_internal: false
  sensor_f6_internal: false
  sensor_f7_internal: false
  sensor_f8_internal: false
  sensor_nir_internal: false
  sensor_clear_internal: false

  sensor_f1_basic_internal: false
  sensor_f2_basic_internal: false
  sensor_fz_basic_internal: false
  sensor_f3_basic_internal: false
  sensor_f4_basic_internal: false
  sensor_fy_basic_internal: false
  sensor_f5_basic_internal: false
  sensor_fxl_basic_internal: false
  sensor_f6_basic_internal: false
  sensor_f7_basic_internal: false
  sensor_f8_basic_internal: false
  sensor_nir_basic_internal: false
  sensor_clear_basic_internal: false

  sensor_illuminance_internal: false
  sensor_irradiance_internal: false
  sensor_irradiance_photopic_internal: false
  sensor_ppfd_internal: false
  sensor_irradiance_par_internal: false
  sensor_color_temperature_internal: false
  sensor_saturation_internal: false
  sensor_rgb_hex_internal: false

sensor:
  - platform: as734x
    type: AS7343
    id: ${sensor_id}
    i2c_id: ${sensor_i2c_id}
    address: ${sensor_address}
    gain: ${sensor_gain}
    atime: ${sensor_atime}
    astep: ${sensor_astep}
    update_interval: ${sensor_update_interval}
    normalize_basic_counts: ${sensor_normalize_basic_counts}

    calibration:
      glass_attenuation_factor: ${sensor_glass_attenuation_factor}
```

The package was instantiated with all spectral raw/basic channels plus NIR exposed:

```yaml
packages:
  spectral_as7343: !include
    file: esphome_packages/spectral_sensor_as7343.yaml
    vars:
      sensor_name: "AS7343"
      sensor_id: as7343
      sensor_i2c_id: bus_a
      sensor_f1_basic_internal: false
      sensor_f2_basic_internal: false
      sensor_fz_basic_internal: false
      sensor_f3_basic_internal: false
      sensor_f4_basic_internal: false
      sensor_fy_basic_internal: false
      sensor_f5_basic_internal: false
      sensor_fxl_basic_internal: false
      sensor_f6_basic_internal: false
      sensor_f7_basic_internal: false
      sensor_f8_basic_internal: false
      sensor_nir_internal: false
```

Observed AS7343 values:
Channel | Raw | Basic
-- | -- | --
380 nm | 192 | 0.075043
415 nm | 769 | 0.299683
445 nm Blue | 908 | 0.345205
480 nm | 491 | 0.186228
515 nm Green | 1286 | 0.477870
555 nm | 2316 | 0.836398
590 nm | 638 | 0.224530
630 nm Red | 2309 | 0.845075
680 nm Deep Red | 2007 | 0.743402
730 nm Far Red | 779 | 0.281218
910 nm NIR | 119 | 0.041118
NIR | 94 | 0.035797
Clear | 2717 | 1.000000

Derived values:
- Saturation: 15.2%
- Illuminance: 43,236 lx
- Irradiance: 127.71 W/m²
- Irradiance Photopic: 63.30 W/m²
- PAR: 135.73 W/m²
- PPFD: 622.1 µmol/m²/s
- CCT: 3,844 K
- RGB Hex: `#c49c7c`

The sensor initialized and is producing readings successfully with PR #18012.

I also noticed this compiler warning during the build:

```text
../src/esphome/components/as734x/as734x.cpp:37:13: warning:
'const char* esphome::as734x::{anonymous}::model_name(esphome::as734x::Model)'
defined but not used [-Wunused-function]
```

Apart from this unused-function compiler warning, the build completed and the AS7343 is working correctly.

I use `JGS1 Quartz Glass` as a cover for the sensor.

latonita, 2026-09-16:
While PRs are still under review here is HASS chart i've made based on work by [goatboynz](https://github.com/goatboynz)

Here is the link https://github.com/latonita/HA-par-spectrum-card  - it is not enough to show basic counts - they are not spectrum. 
I've added spectrum reconstruction as per official docs. 

Example 1. halogen lamp (black body)
[image]

Example 2. 3000K LED bulb (nice blue peak)
[image]