esphome/feature-requests #2599: Bluetooth proxy support for ESP32-C6

reinisb, 2024-02-08:
**Describe the problem you have/What new integration you would like**
I'm trying to install ESPHome bluetooth proxy on my ESP32-C6 board, but it fails stating that ESP32-C6 is not supported.

[image]

**Please describe your use case for this integration and alternatives you've tried:**
I would like to install ESPHome bluetooth proxy on my ESP32-C6 board.

**Additional context**
No additional context

andrewseago, 2024-02-12:
I also would appreciate C6 support.

randybb, 2024-02-12:
The latest esphome is runningh on ESP-IDF v4.4.6, C6 is supported from ESP-IDF version 5.1.To switch to 5.1 will take a while, then the template will be updated.

reinisb, 2024-02-12:
> The latest esphome is runningh on ESP-IDF v4.4.6, C6 is supported from ESP-IDF version 5.1.To switch to 5.1 will take a while, then the template will be updated.

Good to know, thanks for sharing some context. Is there an issue where the work to switch esphome to 5.1 is tracked?

randybb, 2024-02-12:
We need to make everything compatible with 5.0 first, then we will go to 5.1, so this will take some time. There is an old discussion about this on [discord, ESP-IDF 5.0.0](https://discord.com/channels/429907082951524364/1076282397903048725), some initial support for 5.0 has been already completed, now is some work done on a new testing framework.
For now you can only try to switch platformio version and esp-idf manually (check https://github.com/esphome/feature-requests/issues/2176 how to override it and which version has been tested by other people) and something might work, but without an official esphome support, so please don't create issues that something doesn't work.

Hedda, 2024-08-21:
FYI, some other project developers (including Tasmota's lead developer) have decided to fork Platformio's platform-espressif32 library due to lack of official PlatformIO development because of this conflict that they PlatformIO have with Espressif that have now made them even refuse to review and merge community patches to Platformio's platform-espressif32 library:

* https://github.com/esphome/esphome/discussions/7334

> _**Since it is now clear that there will be no official support for Arduino core 3.0.x from Platformio team, (Issue No. 1225 in repo https://github.com/platformio/platform-espressif32/issues) i decided to fork the needed repo(s) and build a community version to support core 3.0.x (starting with core 3.0.3)**_
>
> (_Originally posted by @Jason2866 in https://github.com/espressif/arduino-esp32/discussions/10039_)

Thus if you want to use Espressif Arduino 3.0.x and IDF 5.1.x with newer ESP32 SoCs like ESP32-C6 and ESP32-H2 then you might need to consider using pioarduino's fork of the platform-espressif32 instead of the "official" platform-espressif32 from Platformio.

* https://github.com/pioarduino/platform-espressif32

For additional information, please refer to these GitHub links:

* https://github.com/espressif/arduino-esp32/discussions/10039
* https://github.com/platformio/platform-espressif32/issues/1225
* https://github.com/espressif/arduino-esp32/pull/8606

ssalahi, 2024-10-15:
Any news on support of Bluetooth Proxy for ESP32-C6?

kabakaev, 2025-01-16:
To use an ESP32-C6 as a Bluetooth Low Energy (BLE) proxy, install ESPHome with the newest version of the ESP-IDF framework. Here is a tested example:
https://gist.github.com/kabakaev/759a3292112225db50b98e2254598793

ssalahi, 2025-01-18:
Thank you @kabakaev, the yaml and instructions worked!
The C6 board is super stable compared to the generic version. It can be because of ESP-IDF or the WiFi 6 or both.

horza22, 2025-02-03:
> To use an ESP32-C6 as a Bluetooth Low Energy (BLE) proxy, install ESPHome with the newest version of the ESP-IDF framework. Here is a tested example: https://gist.github.com/kabakaev/759a3292112225db50b98e2254598793

Really appreciated [kabakaev](https://github.com/kabakaev) got this working thanks to this!

ajoe, 2025-04-04:
> Thank you [@kabakaev](https://github.com/kabakaev), the yaml and instructions worked! The C6 board is super stable compared to the generic version. It can be because of ESP-IDF or the WiFi 6 or both.

Hi. I get an error: that the API Token is not base64 encoded. If I encode it with base64, the next error is that the token is longer than 32 byte length. Do you have an idea why this happens?
Homeassistant version: "2025.3.4".

ardeus-ua, 2025-04-19:
In my case it perfectly worked with esp-idf 5.5.0 and latest pioarduino release for 5.5.0

ardeus-ua, 2025-04-24:
> Can you post how you have done it? Or reference a description?

esp32:
  board: esp32-c6-devkitc-1
  flash_size: 8MB
  framework:
    type: esp-idf
    version: 5.5.0
    release: "250407"
    platform_version: 54.03.20