esphome/feature-requests #2600: rp2040_pio_led_strip add support for WS2805 RGBCCT LED strips

Kiendeleo, 2024-02-09:
**Describe the problem you have/What new integration you would like**
 Add support for RGBCCT LED Strips, such as the WS2805 LED 

**Please describe your use case for this integration and alternatives you've tried:**
 I personally want to use this for lighting in my office so I can dim the lighting and use the tunable lights to move to a warmer light after sunset.  I also want to use this for permanent holiday lights that can also be used to illuminate areas around the house. 

**Additional context**

Example of the LED Strip1: https://www.superlightingled.com/ws2805-dual-signal-5in1-rgbww-addressable-rgb-tunable-white-led-strip-p-6048.html
Example of the LED Strip2: https://www.btf-lighting.com/collections/ws2805-rgbcct-24v/products/ws2805-pixels-led-strip-light-rgbcct-5in1-60leds-m-addressable-dual-signal-3000k-6500k

agl1tch3dpir8, 2024-09-03:
I'm not the coding type, but am definitely willing to purchase hardware to contribute to the development of this feature!

jtulak, 2025-02-22:
It looks like the only led strip platform where this would make sense and which is working with the current framework version is `ESP32 RMT`. `SPI LED Strip` makes no sense as WS2805 is clockless and and `FastLED` is broken, with last usable version in framework `2.7.4`:
https://github.com/FastLED/FastLED/issues/1322
https://github.com/FastLED/FastLED/issues/1264

So (imo) there is no chance of this getting a wide support until FastLED is fixed. :-(

jtulak, 2025-03-19:
Also, NeoPixelBus includes support for WS2805 since the middle of 2024, release [2.7.9](https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.9)
MR: https://github.com/Makuna/NeoPixelBus/issues/764

I have no idea what version of NeoPixelBus ESPHome is using, but it seems not new enough

Related feature request for RMT implementation: #3063

jtulak, 2025-03-19:
Also, WLED supports WS2805, so it is a workaround, if we can call it that way: https://kno.wled.ge/basics/compatible-led-strips/#addressable-led-strips