esphome/feature-requests #2603: Tuya MCU refresh timer

Bl00d-B0b, 2024-02-13:
Hello,

I use several MOES thermostat. with original software I use default calibration "-3", but when I install esphome based on bk7231t I must increase calibration up to "-6", As I understand, it's happens, because TUYA MCU receive  data every second from serial interface.

is it possible to read specify refresh rate to read data from tuya MCU?

Ed

ssieb, 2024-02-13:
What do you mean by calibration?  esphome doesn't poll it, the MCU sends data when it wants and esphome sends data only when changing a setting.

Bl00d-B0b, 2024-02-13:
All MOES thermostat have internal calibration:

[image]

So, with original firmware I Leave default, and temperature is accurate, but when I install ESPhome, when I need to add additional "-3"... When I power on device temperature accurate, but when I wait 2-3 mins, then it's not accurate and requires calibration, as i mentioned before.

ssieb, 2024-02-13:
Maybe it's the ESP that's causing the heat.

Bl00d-B0b, 2024-02-13:
That's is my assumption (I use bk7231t not ESP). But when I use original firmware (tuya cloud connected), then I don't see this behavior. I don't know how often tuya mcu send updates in original firmware. 
P.S. External nfc sensor shows correct temperature with both firmware's.

ssieb, 2024-02-13:
It's not the Tuya MCU.  esphome doesn't change how that works.  It's more likely the BK CPU if anything.  Maybe esphome runs it harder.

Bl00d-B0b, 2024-02-13:
Anything ideas how i can troubleshoot this? 

P.S. also I notice, than thermostat after several days hung, and only hardware power restart help them, soft restart give error initializing tuya mcu...

Ed

Bl00d-B0b, 2024-02-26:
Hello,

Looks like the MCU update interval is required feature!!!, I read some whitepapers, and it's not recommended to read temperature sensor every second, then sensor shows wrong temperature, it's not a problem with ESP heat. CPU is quite far away.

Is it possible to implement a possibility to read temperature sensor with timeout option.

And also, another problem still exist even with new firmware. Chip is halt, only power restart helps. looks like need implementation with reset of MCU??? 

Ed

Bl00d-B0b, 2024-02-28:
Hello,

I got another tuya MCU thermostat, which is based on cb3s chip. Successfully use tuya-cloudcutter, and can give help with Datapoints:
Datapoint 1: switch (value: ON)		- on/off
Datapoint 2: enum (value: 1)		- manual/scheduler
Datapoint 3: enum (value: 1)		- ??? R/O
Datapoint 24: int value (value: 235)    - Current Temperature R/O
Datapoint 36: enum (value: 0)           - Heating/Idle R/O
Datapoint 101: int value (value: 0)	- External Temperature R/O
Datapoint 16: int value (value: 225)	- Target Temperature
Datapoint 18: int value (value: 250)	- Max. Set Temp
Datapoint 19: int value (value: 250)	- Max. Set Temp
Datapoint 20: int value (value: 100)	- Min. Set Temp
Datapoint 26: int value (value: 100)	- Min. Set Temp
Datapoint 27: int value (value: -5)     - Temperature Compensation
Datapoint 40: switch (value: OFF)	- Lock
Datapoint 71: raw 			- calendar walues
Datapoint 102: enum (value: 0)		- Int/Ext/Int&Ext
Datapoint 103: int value (value: 1)	- Deadzone Temperature
Datapoint 104: int value (value: 450)	- High Temperature Protection
Product: '{"p":"l26ggao46s756sno","v":"1.0.0","m":2}'

But still see the same behavior:
1) when original firmware, default temperature calibration is -3, and show correct temperature in the room, but when I install esphome, then i need to setup -6 colebration.
2) when power on, device do not receive time via ntp, need to have additional restart
3) receiving always messages: Component tuya took a long time for an operation (0.07 s).
4) but this device never hangs yet (3 days)

maybe someone can implement tuya refresh timeout????
Ed

ssieb, 2024-02-28:
I already told you that the "tuya refresh timeout" isn't a thing.

Bl00d-B0b, 2024-03-11:
OK, maybe then it's possible to add a **deep_sleep** support for bk72xx boards?

Bl00d-B0b, 2024-03-12:
I also see the similar behavior on Wi-Fi temperature/humidity sensor on CBU board,  they are useless, as a battery live only 1-2 days with esphome., comparing 3 month with original firmware.

Bl00d-B0b, 2024-03-27:
Hello,

Still problem exist and not solved. thermostat with Tuya MCU hung (15-30 hours working) and temperature sensor start to show increased temperature +3/+4 degree after working about 15 mins. Original Firmware does not have those issues.

I am assuming, it's because esphome Tuya read ir every 1-2 seconds. 

Is it possible to disable and enable tuya mcu or serial interface based on schedule, for example every 1 minute?

Ed