esphome/feature-requests #2604: Better error message when flashing with improper permissions

OdileVidrine, 2024-02-13:
**Describe the problem you have/What new integration you would like**
On Linux, when attempting to flash a device, depending on the tty permissions, it might fail to connect and give the error: `ERROR Running command failed: Could not open /dev/ttyUSB0, the port doesn't exist`. The tty required the dialout group, and my user was not a member of the dialout group.
I would like this error message to be more specific, as it doesn't indicate that the port could exist but I do not have the correct permissions to see it. A better error message would tell me to check the user privileges and give the command to add the user to dialout.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

flaviut, 2024-02-13:
The implementation here would likely be to make a new function in https://github.com/esphome/esphome/blob/7baf091d47cbfb426918a0a836bd4dda8c8eb93b/esphome/__main__.py#L300 that checks:

1. if we're running on a system with unix-style permissions, validate that we can open the selected serial port. If we can, we're done.
2. if we can't, and g+rw is set, and the user is not in the tty file's group, and the tty file's group is not root, print a message explaining how to add the user to the group and log out & back in/restart the service.
3. otherwise, print a generic message about not being able to open the serial port and some details about ownership/permissions of the port and of the current user.

it would make sense the run this function before either `upload_program` or `show_logs`. The initial version of this function probably shouldn't gate the rest of the functions, in case there are bugs or edge cases that were missed.

randybb, 2024-02-14:
something similar has been added recently to [esp-web-tools](https://github.com/esphome/esp-web-tools/commit/21c23206d088fa1bd663907c4226c6874eba977a)