esphome/feature-requests #2607: Add support in Home Assistant ESPHome Addon for ESPHome dashboard running on a separate computer on the user's LAN

philpownall, 2024-02-15:
**Describe the problem you have/What new integration you would like**

I presently use esphome in a manual install on a desktop machine which is much more powerful than the machine that I run Home Assistant Operating System on.  For example, platformio and esphome "installs" run about 10 times faster on my 4-core desktop than on my HA server (a beelink mini pc) - e.g. seconds rather than minutes.  

This is a useful alternative,  but I cannot edit and install from the Home Assistant Addon, as the dashboard is actually on another computer - which I access via a browser using the LAN IP address of the manual dashboard install running on that computer, and all of the yaml files, build directories, and the (huge) platformio installs are on the desktop computer as well.

**Please describe your use case for this integration and alternatives you've tried:**

The ESPHome Home Assistant Addon INSTALL (compile, link,and OTA firmware upload) is too slow and takes up too much space on a typical Home Assistant server; it is much faster on a separate desktop.

The problem, if this is even possible, is that there is no "configuration" option in the Home Assistant ESPHome Addon to specify the IP address of the computer that is actually running the ESPHome dashboard.

The VISIT link for a node in the ESPHOME Addon dashboard in Home Assistant works just fine to access a node web page, and the devices link to the Home Assistant api just fine.

**Additional context**

I realize I could just remove the ESPHome Addon in Home Assistant, but I suspect that adding a configuration option to enter the ip address of the ESPHome dashboard might actually be quite easy.  It doesn't have to be enabled by default.

ssieb, 2024-02-15:
The addon **is** the dashboard.  Pointing it somewhere else doesn't make sense.  Do you know that you could run the dashboard on your local computer since that's where you have esphome installed?

philpownall, 2024-02-15:
It does make sense.  I have esphome running on my home assistant server AND
on a pc on my LAN.  I have the esphome dashboard also running on both
devices.

When I open the ESPHome dashboard from within Home Assistant, it points to
the server
, and any installs done from here will execute on the Home Assistant Server.

When I open the esphome dashboard on my pc, the installs are done on the
pc, and the compile and OTA is much faster.

So, why can I not point esphome dashboard at my pc rather than the Home
assistant server?

Regards,
Phil.

ssieb, 2024-02-15:
I don't understand what you mean by "pointing".  If you want the dashboard running on HA, then go to that page.  If you want the one running on your computer, go to that one.  There's nothing that points at anything else.  Make a link in HA that points to your computer dashboard if you want.  The dashboard can't run builds on another computer.

philpownall, 2024-02-15:
Thank you Samuel for your quick response.

I am aware that I can open the esphome dashboard on my pc (and run builds)
by opening a link to the esphome dashboard on my pc from within Chrome or
Firefox.  That's what I do now.

Perhaps it is a little unclear because there are Lovelace dashboards and an
ESPHome dashboard running in a docker container on the Home Assistant
server.  What I mean by "pointing" is that when you click on the ESPHome
menu link inside Home Assistant it opens up a Lovelace view of the
server-based esphome dashboard.

You say that the dashboard (which one?) can't run builds on another
computer, but that's exactly what I am doing - I open a link to the esphome
dashboard on my pc in Firefox from my laptop, and the yaml files,
platformio, and esphome are all on my pc downstairs in my office.  And if I
click on the VISIT link from within the Home Assistant ESPHome dashboard,
it opens the node web page (esphomeX.local) in my browser.  All fine.

You say that I can make a link in HA that points to my pc, I'll head in
that direction (can you define a Lovelace view that opens a browser LAN ip
address?  Awesome) and I can remove the ESPHome addon from my Home
Assistant server.

Regards,
Phil

ssieb, 2024-02-15:
I really don't understand what you're saying.  I think you're still mixing up different things.
The Lovelace dashboard is using the esphome **integration** that's part of HA.  The esphome dashboard where you can compile is the **addon**.  They are completely separate.
The visit link goes directly to the ESP device, not HA and not the esphome dashboard.
If you are compiling on your PC, then you're not using the dashboard from the addon at all.

nagyrobi, 2024-02-16:
@ssieb I think he simply wants to load as an ingress page the ESPHome dashboard of the PC, not the add-on.

randybb, 2024-02-16:
Well, this this is possible - I have a separate server (cube) which is running another two instances of esphome and grafana
[image]

But I think he just wants that link under a device to the esphome dashboard - this is possible only with the addon and only to that which has "Home Assistant Dashboard Integration" enabled
[image]

philpownall, 2024-02-16:
Thanks for pointing me in the right direction everyone - I added a button url tap action to a Dashboard, with a link to the ESPHome dashboard running on my pc (which opens a new browser window), and turned off the "Show in Sidebar" toggle in the ESPHome Home Assistant Addon "Settings -> Addons -> ESPHome Addon -> Info tab " so that the ESPHome entry in the Home Assistant page no longer appears.  This solution even works on my phone from the Companion App.  Now on to figuring out how to change the button icon to the ESPHome svg...

randybb, 2024-02-16:
You can add something to the left panel via [iFrame panel](https://www.home-assistant.io/integrations/panel_iframe/).

Unfocused, 2024-02-29:
Using the dashboard directly is not the only use of the dashboard though. 

You can update ESPHome devices from within Home Assistant itself (added in Home Assistant 2023.2 and ESPHome 2023.2). Each ESPHome device has an update entity, so you can update a device from the device details page in Home Assistant, plus it's listed in Home Assistant's updates page, and you can just add it to a Home Assistant dashboard like any entity to get a nice little update button without ever seeing the ESPHome dashboard. For just updating things for regular maintenance, I find Home Assistant's Updates page far more convenient than going through to several other dashboards (there's why it's there!). 

But that update entity actually uses the ESPHome dashboard to do all the work.

It communicates with the dashboard using the [dashboard-api](https://github.com/esphome/dashboard-api) library, which uses HTTP WebSockets. There's no reason that has to be the dashboard hosted by the Home Assistant ESPHome add-on. Point that at a different IP, and it should just automagically work.

philpownall, 2024-03-12:
Thanks, I've taken Roman's advice and removed the esphome addon from my
home assistant server, and replaced it with a button card url pointing to
my development pc running the esphome dashboard (manually installed).

Actually, I've got two button cards - one for the esphome dashboard pc on
my home lan, and one for my cottage lan pc, with the esphome yaml files
synchronized through github desktop in both locations.  My home assistant
server load has gone down to almost zero, I can happily update code in both
locations, so I'm happy with that, and the builds are much, much faster.

Thanks and regards,

closed 2024-08-17 by nagyrobi