esphome/feature-requests #2610: [HLW8012] Allow the use of the library defining only CF pin

abl20, 2024-02-18:
**Describe the problem you have/What new integration you would like**
Currently, the HLW8012 library forces you to define 3 pins (SEL, CF and CF1) in order to use it.
This library uses pulse_counter function for two pins (CF and CF1). The third pin (SEL) is defined as a normal output, so:

- SEL (change mode) -> digital output.
- CF (current or voltage, depends on SEL value) -> pulse counter input.
- CF1 (power) -> pulse counter input.

When using this library in a ESP32-S3, there is a limitation since ESP32-S3 only has 4 Hardware Pulse Counting Module (PCNT). This means that you cannot use more than two HLW8012 devices (2xCF and 2xCF1) per ESP32-S3 microcontroller .
It is not really necessary to use all pins, since for most applications measuring power is more than enough, and the SEL and CF1 pins do not depend on CF. So you can use CF without even having SEL and CF1 connected.
My proposal is to be able to define only the CF pin, in order to free PCNT channels and be able to use up to 4 HLW8012 devices (or other devices that use PCNT) at the same time in a single ESP32-S3 😊

**Please describe your use case for this integration and alternatives you've tried:**
So far if you define, for example, 4 HLW8012 devices (using esp32-s3) it doesn't give you any error, it only shows the values of the first 2 you define (because of PCNT channels).

One workaround I tried is to use a common CF1 "dummy pin" by setting "allow_other_uses: true" on pin configuration. The code compiles but it does not improve anything.
______________________________
#Device1
    cf1_pin:
      number: 15
      allow_other_uses: true
    cf_pin: 4
#Device2
    cf1_pin:
      number: 15
      allow_other_uses: true
    cf_pin: 5
_______________________________
Before I imagined the assignation would work as follows:
PCNT CH0 = 15
PCNT CH1 = 4
PCNT CH2 = 5
PCNT CH3 = available

But once testing it, I can guarantee that the assignment made by the library is:
PCNT CH0 = 15
PCNT CH1 = 4
PCNT CH2 = 15
PCNT CH3 = 5

So, there is no improvement with this solution, as it is using the same amount of PCNT channels. I would say it only will be possible by modifying the library.

**Additional context**
This is my first time opening a feature request, hope it is clear enough for all of you! Anyway, I will follow this post on a daily basis.
Also hope this is not my last one and start contributing a little more to this great ESPHome world!

ESP32-S3 Technical Reference evidence:
[image]

nagyrobi, 2024-02-20:
When using only the CF ping as a pulse output, you could simply use ESPHome's built-in `pulse_meter` sensor directly.
For example on Lanbon L8 there's a HLW8012 chip with only CF connected to GPIO35 (no SEL and CF1 connected).

```yaml
sensor:
  - platform: pulse_meter
    name: Lanbon Power Consumption
    id: sensor_pulse_meter
    pin: GPIO35
    unit_of_measurement: 'W'
    device_class: power
    state_class: measurement
    internal_filter_mode: PULSE
    accuracy_decimals: 1
    filters:
      - filter_out: nan
      - throttle: 1s
      - multiply: 0.0813287514318442  # Calibration may be needed
```

To calibrate the power values measured by the `pulse_meter` sensor, use an external power meter which is known to make correct measurements, and attach an ohmic load of about 70-100W (an incandescent bulb, or a small heater). In the config, replace the multiply value with 1, and flash the device. Turn on the load and observe the reading on your external power meter and the value reported by the sensor. Your calibrated new multiply value will be external power meter measurement / the value reported.

abl20, 2024-02-21:
> When using only the CF pin only as a pulse output, you could simply use ESPHome's built-in `pulse_meter` sensor directly. For example on Lanbon L8 there's a HLW8012 chip with only CF connected to GPIO35 (no SEL and CF1 connected).
> 
> ```yaml
> sensor:
>   - platform: pulse_meter
>     name: Lanbon Power Consumption
>     id: sensor_pulse_meter
>     pin: GPIO35
>     unit_of_measurement: 'W'
>     device_class: power
>     state_class: measurement
>     internal_filter_mode: PULSE
>     accuracy_decimals: 1
>     filters:
>       - filter_out: nan
>       - throttle: 1s
>       - multiply: 0.0813287514318442  # Calibration may be needed
> ```
> 
> To calibrate the power values measured by the `pulse_meter` sensor, use an external power meter which is known to make correct measurements, and attach an ohmic load of about 70-100W (an incandescent bulb, or a small heater). In the config, replace the multiply value with 1, and flash the device. Turn on the load and observe the reading on your external power meter and the value reported by the sensor. Your calibrated new multiply value will be external power meter measurement / the value reported.
> 
> Note that this is a compromising usage of HLW8012, as low powers of around a few watts generate rare pulses, thus it's inadequate for exact measurements. Eg. you turn off the load but the chip will only notice 10 seconds later...

Hi! Thank you for your comment. I am triying your proposal but checking the log seems like it is not reporting any value. Also I cannot set update_interval when using pulse_meter :(

abl20, 2024-06-27:
Hi, could someone give me a hint to be able to move forward and get what I was commenting? Thanks in advance