esphome/feature-requests #2611: Support SNTP with ESP-IDF version > 5.0

vinhpn96, 2024-02-19:
**Describe the problem you have/What new integration you would like**

Could not build sntp component with ESP-IDF version > 5.0
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

randybb, 2024-02-19:
Sure, this is the one of the reasons why esphome is still on esp-idf 4.4 - a rework of sntp is needed. There is no need for such feature request.

closed 2024-02-19 by randybb

vinhpn96, 2024-02-20:
> Sure, this is the one of the reasons why esphome is still on esp-idf 4.4 - a rework of sntp is needed. There is no need for such feature request.

@randybb 
when is esphome on esp-idf > 5.0 ?

randybb, 2024-02-20:
No time plan, but probably this year. Most of the components are working fine (most of my devices are running on esp-idf 5.1.2), some need to be rewritten. But I am not developer - they know what is the best to do.

vinhpn96, 2024-02-21:
> No time plan, but probably this year. Most of the components are working fine (most of my devices are running on esp-idf 5.1.2), some need to be rewritten. But I am not developer - they know what is the best to do.

Thank you for letting me know