esphome/feature-requests #2612: WireGuard on ESP8266

Iulian-Dragomirescu, 2024-02-19:
**Describe the problem you have/What new integration you would like**

Hi, I want to implement wireguard in ESP8266. I finished configuring WireGuard and it worked very well on the ESP32 and when I had to add it to the "production" board I realized that it is not possible....

**Please describe your use case for this integration and alternatives you've tried:**

Please add this to ESP8266. It is very useful for all users to be able to control from outside the home in different networks. Thank you very much!

**Additional context**

ssieb, 2024-02-19:
Someone is actually working on this right now.

Iulian-Dragomirescu, 2024-02-19:
Cool! Would you have an idea when this update would come out?

ssieb, 2024-02-19:
Whenever they finish it. :man_shrugging:

kvlh, 2024-03-28:
@Iulian-Dragomirescu 
look here
https://github.com/esphome/esphome/pull/6365

SimonFischer04, 2025-01-02:
https://github.com/esphome/esphome/pull/6365 is merged, so I think this can be closed?

closed 2025-01-02 by Iulian-Dragomirescu