esphome/feature-requests #2616: Pin Schema for I2C (I2C over I/O Expander)

travnick, 2024-02-23:
**Describe the problem you have/What new integration you would like**

I'd like to have possibility to use [Pin Schema](https://esphome.io/guides/configuration-types#config-pin-schema) for [I2C configuration](https://esphome.io/components/i2c.html#configuration-variables), the same as it's available for [SPI](https://esphome.io/components/spi#configuration-variables).

**Please describe your use case for this integration and alternatives you've tried:**

It sounds reasonable to use a [MCP23Sxx I/O Expander](https://esphome.io/components/mcp23sxx#mcp23sxx-i-o-expander) (GPIO over SPI), and then I2C over it.

**Additional context**

I'm not sure if it's technically possible at all to have I2C over any I/O Expander (didn't find anything on the internet), but I hope it is.

Unfocused, 2024-02-29:
Do you mean something like this?

https://esphome.io/components/tca9548a

travnick, 2024-02-29:
The documentation says that ESP8266 does not have a hardware I2C, but have two of hardware SPI, so ESPHome has to have software I2C implementation https://nodemcu.readthedocs.io/en/latest/modules/i2c/. Am I right?

My point is to have only hardware SPI pins occupied (with software I2C on top of it) and have other ESP8266 pins free to do other stuff.

Unfocused, 2024-03-07:
(This isn't my usual area, I'm just trying to provide some context and help where possible)

Sorry, yes, you are right - ESP8266 indeed has no hardware I2C, and uses software. Although oddly enough my point is still valid - it only supports one I2C bus. That is (was) due a limitation in the library used (Wire.h). But that changed recently with this PR providing a workaround (not in a release version yet):

- https://github.com/esphome/esphome/pull/6145

I'm assuming Wire.h (and it's API) was chosen because it's so commonly used - it means automatic support for a vast number of Arduino libraries. I mention that because unfortunately Wire.h has some significant limitations that can't be fixed without breaking things.

One of those limitations is that Wire.h only supports internal pins. It would need to be replaced with another library in order to use ESPHome's concept of pins (which is what allows using pins on multiplexers as though they were normal internal pins in most cases - like using Pins Schemas for SPI).

I completely agree this would be very nice to have. But AFAICT the easy way to fix this would break a bunch of stuff. And the hard way is... hard.