esphome/feature-requests #2619: Volume parameter in the speaker component

Instg1, 2024-02-24:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Instg1, 2024-02-24:
> **Describe the problem you have/What new integration you would like**
> 
> **Please describe your use case for this integration and alternatives you've tried:**
> 
> **Additional context**

VivantSenior, 2024-03-16:
I agree, adding ability to change volume in the [speaker component](https://esphome.io/components/speaker/index.html) would be great when using [Onju Voice Satelite](https://github.com/tetele/onju-voice-satellite) (hardware by [@justLV](https://github.com/justLV) ([Onju Voice](https://github.com/justLV/onju-voice)), config by [@tetele](https://github.com/tetele)). Example configuration can be found in [onju-voice-microwakeword.yaml](https://github.com/tetele/onju-voice-satellite/blob/main/esphome/onju-voice-microwakeword.yaml). More information here: [#12](https://github.com/tetele/onju-voice-satellite/issues/12#issuecomment-2001884965).

Hedda, 2024-08-15:
FYI, esphome developers are experimenting with volume commands and much more in the related experimental voice-kit repo:

* https://github.com/esphome/voice-kit