esphome/feature-requests #2629: I2C Templerature LM75 (and derivates)

petrkr, 2024-02-27:
**Describe the problem you have/What new integration you would like**
I have PLC board which use this LM75 i2c temperature sensor and I would like to use it. Basically this i2c sensor is very cheap and for some orientation temperature it is fair enough and since there are already other devices on i2c bus so It can be used.

**Please describe your use case for this integration and alternatives you've tried:**
Use case is have temperature. Alternative is custom firmware / micropython or Tasmota, which this driver has.

**Additional context**
I found nice implementation from Arduino IDE https://github.com/jeremycole/Temperature_LM75_Derived Do not know structure of ESPHome and do not know if can be used as-is or must be rewritten to esphome devices tree.

Regards Petr

sbrk-r, 2024-03-01:
It's possible to use custom sensor for this, it works ok. Very strange it isn't part of the base package though, when less common sensors are.

sbrk-r, 2024-03-09:
Today I thought I would write a component for LM75 and I noticed there is a TMP102 integration which is largely compatible. It doesn't hit the config register so you only get 9 bit results (like original LM75, so only 0.5 degree C resolution). I'm testing with TMP75 which supports 12 bit. (0.0625 degree).

TL;DR if you have LM75 and just need the temperature it seems TMP102 integration will work

petrkr, 2024-03-09:
Let me check, for orientation it would be enough. Later I can check also support. But for now I need it as proof-of-concept that board will work and can be used with ESPHome. It is this one https://www.octopuslab.cz/plcshield1/

sbrk-r, 2024-03-09:
I will look through a few more xx75 datasheets and try and write something that accommodates more features (like setting the thermostat/alarm output). I've never put a PR for this before, it will be a little learning experience.

I think it probably makes the most sense to have it default to base lm75 features, and have options to use descendants with more bits, features, etc.