esphome/feature-requests #2630: Tesla vehicle command over BLE

mongojarle, 2024-02-28:
tes
**Describe the problem you have/What new integration you would like**

Support for local control of Tesla vehicle. Send commands to the car (i.e unlock/lock, start charge, set charge amps).
Teslas repo: https://github.com/teslamotors/vehicle-command (everything below "Using the HTTP proxy" is not necessary for local BLE control).
Commands over BLE: https://github.com/teslamotors/vehicle-command/tree/main/cmd/tesla-control 

**Please describe your use case for this integration and alternatives you've tried:**

ESP in garage, connected to vehicle over BLE as well as local wifi. Sending commands to ESP over MQTT/HA to start/stop charging and set charge amp.
Was using Tesla "node" in nodered earlier, but Tesla changed their API, so it no longer works, with little hope for fixing (https://github.com/onokje/node-red-contrib-tesla).
Local control over BLE also bypasses the requirements for internet and the use of Teslas API.

**Additional context**

Gathering information from the car from Teslas API is not as strict as controlling the car. So getting info from the car over BLE is not necessary, or possible I think.
Teslamate still works for logging car, also this is a plus, since logging also works on trips, outside of BLE/wifi range.
Needs to control the car elsewhere than garage/BLE coverage may just as well be done through the Tesla app.

MrBLJ, 2024-06-04:
> Was using Tesla "node" in nodered earlier, but Tesla changed their API, so it no longer works

That's definitely correct. We can make use of a proxy to keep sending commands but there's a major issue : Tesla introduced a cap on the number of commands that can be sent per car each day. For those of us who'd like to optimize their solar power production by making use of commands like start/stop charging and set charge amp, this is a big limitation.

mongojarle, 2024-06-04:
> Tesla introduced a cap on the number of commands that can be sent per car each day.

Another good reason to why just use BLE, as it is local directly to the car, not through any of Teslas servers.

Here is a project using python: https://github.com/kaedenbrinkman/PyTeslaBLE

nagyrobi, 2024-06-04:
Someone should just make a Home Assistant custom component out of it, and just use ESPHome's `bt_proxy`.

MrBLJ, 2024-06-04:
> Someone should just make a Home Assistant custom component out of it, and just use ESPHome's `bt_proxy`.

Someone should indeed. Question is who's capable and willing to ? I'm currently reading on the topic, [and found some interesting resources on the subject.](https://shankar-k.medium.com/tesla-developer-api-guide-ble-key-pair-auth-and-vehicle-commands-part-3-485e4a357e7d)

The question now is how do we convert this into an esphome module/integration (not even sure how to call it) ? For now, it's beyond my knowledge. Not sure ChatGPT will know much about this either.

MrBLJ, 2024-06-04:
[And it looks like that someone is working on the topic already.](https://github.com/pmdroid/tesla-ble/issues/1)

WeiyiWang-Cochlear, 2024-06-12:
Definitely would like to have this, for exactly the same reason of controlling charging amps to make use of solar power. 
The new fleet API has ridiculous rate limit that can't be used for this purpose.

MrBLJ, 2024-06-12:
> Definitely would like to have this, for exactly the same reason of controlling charging amps to make use of solar power. The new fleet API has ridiculous rate limit that can't be used for this purpose.

This is quite fresh and new, but it's working. https://github.com/iainbullock/tesla_ble_mqtt_docker

MrBLJ, 2024-06-12:
> > > Definitely would like to have this, for exactly the same reason of controlling charging amps to make use of solar power. The new fleet API has ridiculous rate limit that can't be used for this purpose.
> > 
> > 
> > This is quite fresh and new, but it's working. https://github.com/iainbullock/tesla_ble_mqtt_docker
> 
> Wow! Nice find! I was hoping to get it work with an ESP32 (for its low power consumption) and place it around the car, but a r-Pi is fine as well, at least for now. Did you have it running already?

Yes it's up and running. The number of commands is limited and there are minor issues for now, but I guess the hardest part is done (adding the key to the vehicule was painless, and commands are received well).

I wrote a tutorial to set this up (doc is clear but I think some users will have difficulties), use your browser to translate to english: https://forum.hacf.fr/t/envoyer-des-commandes-a-sa-tesla-via-mqtt-ble/43348

WeiyiWang-Cochlear, 2024-06-12:
> Yes it's up and running. The number of commands is limited and there are minor issues for now, but I guess the hardest part is done (adding the key to the vehicule was painless, and commands are [...]

Thanks! I'll give it a try later today after work.
Should be rather straight forward as I understood. I managed already earlier to send commands to the car from my Mac with the car in it's bluetooth range. 

What do you actually mean by "the number of commands is limited"?

MrBLJ, 2024-06-12:
> Thanks! I'll give it a try later today after work. Should be rather straight forward as I understood. I managed already earlier to send commands to the car from my Mac with the car in it's bluetooth [...]

At this stage, you can only :
- Start/Stop charging
- Open/Close the trunk
- Set charge current (min 5A)

There's no limit on the amount of commands you can send (poor choice of words in my previous message sorry).

More commands can be added, but from what I've understood the author is on holiday for now.

As for the charge current issue, it's under scrutiny. I'm confident it'll be fixed.

WeiyiWang-Cochlear, 2024-06-12:
Aha got it. Setting charging current is my urgent need at the moment so that's fine. 
Adding supports to other commands wouldn't be complex either I assume, considering that the Tesla BLE api does support a wide range of commands.

mongojarle, 2024-06-12:
Some WIP here as well: https://github.com/robin-thoni/tesla-ble-esphome

hitnrun30, 2024-06-15:
> Some WIP here as well: https://github.com/robin-thoni/tesla-ble-esphome

So for this to work you just need an ESP 32 whatever setup in ESPhome, I mean it does make it local, so that is good, my biggest thing I want to do is know if it is charging and set the temperature to go on in the morning or when I'm leaving.  Is that possible with this eventually?

raphmur, 2024-06-24:
Hi there
We (team of 3) are also working on a custom ha Addon: https://github.com/tesla-local-control/tesla-local-control-addon

We will merge with this one as well https://github.com/iainbullock/tesla_ble_mqtt_docker

Today the HA Addon is working, relatively stable, ans brings you command buttons to operate your solar charge, and all the others (AC, windows, trunk, honk, ...). It also implements a presence scanner. But not capability yet to read car values.

This is fairly new so moving quickly. It can be used as is in home assistant using built in bluetooth (eg. on RPi) or can be deployed as standalone docker. It then populates MQTT entities.

Please come and comment there to make thing progress. In the next steps we have: bluetooth proxy (esp)... It would make a lot of sense to empack different variants in the same place (python variants, esp standalone, ...)

mr-p666, 2024-07-05:
That's awesome! :+1:

raphmur, 2024-07-05:
Super nice yes 👍

HayderHekmat, 2024-07-07:
It will be nice if there some instruction how to use EspHome add-on in HA to program an ESP. 
@robin-thoni: did you use the evcc or HA with your device?

robin-thoni, 2024-07-07:
> It will be nice if there some instruction how to use EspHome add-on in HA to program an ESP.

That's out of scope. There's already the [official doc](https://esphome.io/guides/getting_started_hassio) for this.

> did you use the evcc or HA with your device?

I'm using HA

HayderHekmat, 2024-07-14:
@andig and @robin-thoni,
I have successfully installed BLE-esphome ( https://github.com/robin-thoni/tesla-ble-esphome) . I configured also the evcc but, I got always those error.

ERROR 2024/07/14 17:34:09 max charge current 4A: Post "http://192.168.178.xxx:8080/api/1/vehicles/XX/command/set_charging_amps": dial tcp 192.168.178.xx:8080: connect: connection refused

I stacked first with similar issue in #14616 and after many restart. 
Then the evcc started to try to access the BLE proxy.
But this issue happening many times of not considering the proxy command.

 ping 192.168.178.xx

Pinging 192.168.178.xx with 32 bytes of data:
Reply from 192.168.178.xx: bytes=32 time=274ms TTL=255
Reply from 192.168.178.xx: bytes=32 time=80ms TTL=255
Reply from 192.168.178.xx: bytes=32 time=90ms TTL=255
Reply from 192.168.178.xx: bytes=32 time=94ms TTL=255

robin-thoni, 2024-07-14:
I'm not sure what you're trying to do, since I've never used evcc. The only thing I can tell you is that setting the max current charge limit is currently not supported in tesla-ble-esphome, as it's not part of VCSEC, which is the only supported component at the moment

yoziru, 2024-07-16:
I've added support for `UniversalMessage` and both the VSSEC + Infotainment domain in https://github.com/yoziru/tesla-ble
And then integrated into an ESPHome project at https://github.com/yoziru/esphome-tesla-ble
Work-in-progress but works 🤷

robin-thoni, 2024-07-17:
Good job! Can't wait to test this!

LubosD, 2024-08-31:
Q for those using RPi with a Tesla: how well does it work for you?

I am able to hang and crash a RPi with the BLE traffic, about once per day or two. It's not really usable. Happening with both RPi4 and Zero 2 W.

I'm now looking into using ESP32, because that one can at least restart if it hangs...

burschi87, 2024-09-05:
when it works it works really well. But I have to restart the raspberry once per week because of error 429. I don't know whether it is evcc or the ble proxy

BM183, 2024-10-21:
Hi all, I'm interested in getting this working with an ESP32/ bluetooth proxy.
I've seen these projects exist:
https://github.com/yoziru/esphome-tesla-ble
or https://github.com/robin-thoni/tesla-ble-esphome
Is that possible to combine something like this with EVCC and its tesla commandproxy? Thanks!