esphome/feature-requests #2633: Add support ST7701 RGB SPI display

tiimsvk, 2024-03-01:
**Describe the problem you have/What new integration you would like**
Please add support TFT capacitive touch display driver GT911 + ST7701 RGB+SPI 40pin

**Please describe your use case for this integration and alternatives you've tried:**
link on aliexpress: https://www.aliexpress.com/item/1005006302251538.html
support files: http://pan.jczn1688.com/directlink/1/ESP32%20module/4.0inch_ESP32-4848S040.zip?spm=a2g0o.detail.1000023.1.606ePmrUPmrU2K&file=4.0inch_ESP32-4848S040.zip
info on HA community: https://community.home-assistant.io/t/potential-nspanel-alternative-anyone-have-more-information-on-this-device/655236

Thanks

**Additional context**

tiimsvk, 2024-03-01:
Arduino code
[Arduino_ST7701_RGBPanel.zip](https://github.com/esphome/feature-requests/files/14461107/Arduino_ST7701_RGBPanel.zip)

nielsnl68, 2024-03-01:
There is already an PR that should work with the ST7701 from @clydebarrow.

tiimsvk, 2024-03-01:
Yes this: https://github.com/esphome/esphome/pull/5872
just trying to configure the pins

Derek-K, 2024-03-22:
I have been looking for ``4.0inch_ESP32-4848S040.zip`` and the link no longer works. Is it possible to reupload it somewhere for download? thanks.

tiimsvk, 2024-03-22:
> I have been looking for `4.0inch_ESP32-4848S040.zip` and the link no longer works. Is it possible to reupload it somewhere for download? thanks.

This link is ok: http://pan.jczn1688.com/directlink/1/ESP32%20module/4.0inch_ESP32-4848S040.zip

closed 2024-03-22 by clydebarrow

Derek-K, 2024-03-23:
> > I have been looking for `4.0inch_ESP32-4848S040.zip` and the link no longer works. Is it possible to reupload it somewhere for download? thanks.
> 
> This link is ok: http://pan.jczn1688.com/directlink/1/ESP32%20module/4.0inch_ESP32-4848S040.zip

I know this URL but gives 404 error

clydebarrow, 2024-03-23:
Works for me.

agillis, 2024-03-27:
@clydebarrow Do you have a configuration example for this screen?  I tried the code below but it didn't work. All the info you should need is here.

https://github.com/HASwitchPlate/openHASP/blob/master/user_setups/esp32s3/esp32-s3-4848S040.ini

`display:
  - platform: st7701s
    auto_clear_enabled: false
    update_interval: never
    spi_mode: MODE3
    color_order: RGB
    dimensions:
      width: 480
      height: 480
    invert_colors: true
    transform:
      mirror_x: true
      mirror_y: true
    cs_pin: 39
    de_pin: 18
    hsync_pin: 16
    vsync_pin: 17
    pclk_pin: 21
    init_sequence:
      - 1 # select canned init sequence number 1
      - [ 0xE0, 0x1F ]  # Set sunlight readable enhancement
    data_pins:
      red:
        - 4         #r1
        - 3         #r2
        - 2         #r3
        - 1         #r4
        - 0         #r5
      green:
        - 10        #g0
        - 9         #g1
        - 8         #g2
        - 7         #g3
        - 6         #g4
        - 5         #g5
      blue:
        - 15        #b1
        - 14        #b2
        - 13        #b3
        - 12        #b4
        - 11        #b5`

tiimsvk, 2024-03-28:
what device do you have?

clydebarrow, 2024-03-28:
To embed yaml or other code enclose it in with three back ticks = ` ``` `

You didn't show the rest of your yaml - did you set up the SPI? Did you turn on the backlight?

agillis, 2024-03-28:
This code complies fine and boot up with no errors. The backlight does not come on and the screen is black.

```
esphome:
  name: esp32s3
  friendly_name: esp32s3

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
  flash_size: 16MB    

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "75BPcrp4DpO6QMWcvUiQuZMml5ePMk7AidkCzTWx9VY="

ota:
  password: "b94b562281f06b5162bfe38b47d6f8f5"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32S3 Fallback Hotspot"
    password: "UT27cXnwGaSk"

captive_portal:

font:
  - file: "gfonts://Roboto"
    id: roboto30
    size: 30

output:
  - platform: ledc
    pin: GPIO38
    id: gpio38
    frequency: 100
    
light:
  - platform: monochromatic
    output: gpio38
    name: "Backlight"
    restore_mode: ALWAYS_ON
    

spi:
  clk_pin: 48
  mosi_pin: 47
  miso_pin: 41

display:
  - platform: st7701s
    auto_clear_enabled: false
    update_interval: 5s
    spi_mode: MODE3
    color_order: RGB
    dimensions:
      width: 480
      height: 480
    invert_colors: true
    transform:
      mirror_x: true
      mirror_y: true
    cs_pin: 39
    de_pin: 18
    hsync_pin: 16
    vsync_pin: 17
    pclk_pin: 21
    init_sequence:
      - 1 # select canned init sequence number 1
    data_pins:
      red:
        - 11         #r1
        - 12         #r2
        - 13         #r3
        - 14         #r4
        - 0          #r5
      green:
        - 8         #g0
        - 20        #g1
        - 3         #g2
        - 46        #g3
        - 9         #g4
        - 10        #g5
      blue:
        - 4        #b1
        - 5        #b2
        - 6        #b3
        - 7        #b4
        - 15       #b5
    lambda: |-
      it.print(0, 0, id(roboto30), Color(55, 183, 220), TextAlign::TOP_LEFT, "Hello World!");
`

tiimsvk, 2024-03-28:
On my device final +/- code yaml
```
#-------------------------------------------
# BUG AND REPAIR AND PLANNED
#-------------------------------------------

#-------------------------------------------
# REVISION
#-------------------------------------------
# 2024.03.01 - Fisrt version

#-------------------------------------------
# FEATURES
#-------------------------------------------

#-------------------------------------------
# SUBTITUTIONS - basic definitions and description of the device
#-------------------------------------------
substitutions:
  friendly_name: assitant-voice
  device_name: assitant-voice
  created_by: "StudioTiiM 2024"
  device_description: "Voice assitantn HA"
#-------------------------------------------
# POWER CONSUMPTION 
# whit multimeter
#-------------------------------------------
# Power consumption V
# Booting Peak MCU:         mA
# Normal-  wifi:            mA
# Normal - no wifi:            mA
# Deep sleep:              mA

#-------------------------------------------
# DEFINE PINS AND VARIABILES
# CIRCUIT AND MODULES - basic definitions and description of the device
#-------------------------------------------
 # ----- MCU -----
  # Module: ESP32-S3-WROOM-1U-N16R8
   # Circuit: 
   # LDO:
   # Current per pin: max 40mA (default 20mA) / Cumulative: 1500mA
   # Power Consumption: Peak   / Normal wifi:    / Normal:   / Wifi disable: / DeepSleep:  
   # Specification: Chip Family: ESP32 / Chip Model: ESP32-D0WDQ6 (revision 1) / Number of Cores: 2 /Max CPU Frequency: 240MHz / Has Bluetooth: YES / Has Embedded Flash: NO / Has Factory-Calibrated ADC: YES /MAC Address: C8:C9:A3:C7:C4:F4
   

   # Pinout:
    #                       +-----------------------+
    #                       |  +--+  +--+  +--+--+  |
    #                       |  |  |  |  |  |  |  |  |
    #                       |  |  +--+  +--+  |  |  |
    #                       |                       |
    #                GND --+|                       |+-- GND
    #                3V3 --+|                 |o|   |+-- GPIO01 ADC1 TOUCH
    #                 EN --+|                       |+-- GPIO02 ADC1 TOUCH
    # ADC1 TOUCH  GPIO04 --+|   +----------+        |+-- GPIO43 TXD0
    # ADC1 TOUCH  GPIO05 --+|   | MCU      |        |+-- GPIO44 RXD0
    # ADC1 TOUCH  GPIO06 --+|   |          |        |+-- GPIO42 MTMS
    # ADC1 TOUCH  GPIO07 --+|   |          |        |+-- GPIO41 MTDI
    # ADC2        GPIO15 --+|   |          |        |+-- GPIO40 MTD0
    # ADC2        GPIO16 --+|   +----------+        |+-- GPIO39 MTCK
    # ADC2 TXD1   GPIO17 --+|     +---------+       |+-- GPIO38 
    # ADC2 RXD1   GPIO18 --+|    -| FLASH   |-      |+-- GPIO37 PSRAM
    # ADC1 TOUCH  GPIO08 --+|    -|         |-      |+-- GPIO36 PSRAM
    # ADC2 USB_D- GPIO19 --+|    -|         |-      |+-- GPIO35 PSRAM 
    # ADC2 USB_D+ GPIO20 --+|    -|        °|-      |+-- GPIO0  NOT_USE 
    #                       |     +---------+       |
    #                       +-----------------------+
    #                        + + + + + + + + + + + +
    #                        | | | | | | | | | | | |
[... 487 lines cut ...]
      red:
        - 11         # R1
        - 12         # R2
        - 13         # R3
        - 14         # R4
        - 0          # R5
      green:
        - 8          # G0
        - 20         # G1
        - 3          # G2
        - 46         # G3
        - 9          # G4
        - 10         # G5 
      blue:
        - 4          # B1
        - 5          # B2
        - 6          # B3
        - 7          # B4
        - 15         # B5

    #lambda: !lambda |-
    #  // Draw the image my_image at position [x=0,y=0]
    #  it.image(0, 0, id(esphome_logo));
    #  it.print(240, 240, id(my_font), TextAlign::CENTER,"Hello ESPHome!");
```

agillis, 2024-03-29:
Thanks that worked!

jl-p, 2024-04-06:
Great awsome it works 

when i use this i receive an error on   

on_client_connected:
    - if:
        condition:
          binary_sensor.is_on: status_online
        then:
          - lvgl.widget.show: lvgl_online_status
  on_client_disconnected:
    - if:
        condition:
          binary_sensor.is_off: status_online
        then:
          - lvgl.widget.hide: lvgl_online_status

and after removing this part from the api section it uploads but touch isnt working. is that normal.
I see it recognize the touch in the logger but thats it. maybe its normal and yes i need to make my own thing anyway. but wanna be sure its i didnt forget something which i need to change before creating something and figure out after days that i missed something by using this sketch

Gjuju, 2024-04-15:
Hello, I'd like to try your code too but :

```
#-------------------------------------------
# PACKAGES
#-------------------------------------------
packages:
  lvgl: !include packages/lvgl-package.yaml
```

I don't understand where can i find that lvlg-package.yam file ?
any clue ?

Best regards.

agillis, 2024-04-15:
This demo is pulling in clydebarrow,s LVGL beta code. That file is in his git repo. This is not required to get the screen working.

agillis, 2024-04-15:
This is a stripped down version that gets a basic hello world page going with no dependencies

```
#-------------------------------------------
# PROTOCOLS
#-------------------------------------------
i2c:
  - id: bus_a
    sda: GPIO19
    scl: GPIO45
    #frequency: 100kHz

spi:
  - id: lcd_spi
    clk_pin: GPIO48
    mosi_pin: GPIO47

#-------------------------------------------
# ESPHOME PLATFORM SETTINGS
#-------------------------------------------
esphome:
  name: 480screen
  friendly_name: 480screen
  platformio_options:
    board_build.flash_mode: dio

#-------------------------------------------
# ESP MCU SETTINGS
#-------------------------------------------
esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: esp-idf
    sdkconfig_options:

      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_SPIRAM_FETCH_INSTRUCTIONS: y
      CONFIG_SPIRAM_RODATA: y

psram:
  mode: octal
  speed: 80MHz

logger:
  level: DEBUG

api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

#-------------------------------------------
# SENSORS
#-------------------------------------------
sensor:

#-------------------------------------------
# BINARY SENSORS
#-------------------------------------------
binary_sensor:

#-------------------------------------------
# GT911 touchscreen sensor
#-------------------------------------------
  - platform: gt911
    name: "Home"
    index: 0
   
#-------------------------------------------
# OUTPUTS
#-------------------------------------------
output:
  - platform: ledc
    pin: GPIO38
    id: backlight
    frequency: 100Hz

#-------------------------------------------
# LIGHTS
#-------------------------------------------
light:
  - platform: monochromatic
    output: backlight
    name: Display Backlight
    id: display_backlight
    restore_mode: ALWAYS_ON

# I/O expander. Touchscreen RESET and INTERRUPT are not used.
pca9554:
  - id: p_c_a
    pin_count: 16
    address: 0x20

touchscreen:
  platform: gt911
  #rotation: 180
  transform:
    mirror_x: true
    mirror_y: true
  id: my_touchscreen
  display: st7701_display

  on_touch:
    - logger.log:
          format: Touch at (%d, %d)
          args: [touch.x, touch.y]
    - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
              touch.x,
              touch.y,
              touch.x_raw,
              touch.y_raw
              );
image:
  - file: https://esphome.io/_images/logo.png
    id: esphome_logo
    resize: 200x162
  - file: RGB_test.png
    id: RGB_test
    resize: 480x480
    type: RGB565
   
    
font:
  - file: "gfonts://Roboto"
    id: roboto20
    size: 20
    extras:
      - file: 'materialdesignicons-webfont.ttf'
        glyphs: [
          "\U000F004B",
          "\U0000f0ed",
          "\U000F006E",
          "\U000F012C",
          "\U000F179B",
          "\U000F0748",
          "\U000F1A1B",
          "\U000F02DC",
          "\U000F0A02",
          "\U000F035F",
          "\U000F0156",
          "\U000F0C5F", 
          "\U000f0084",
          "\U000f0091",
          ]

display:
  - platform: st7701s
    id: st7701_display
    #rotation: 180
    update_interval: 10s
    auto_clear_enabled: True
    data_rate: 8MHz
    spi_mode: MODE3
    color_order: RGB
    dimensions:
      width: 480
      height: 480
    invert_colors: false
    transform:
      mirror_x: true
      mirror_y: true
    cs_pin: 39
      # reset not defined 
    reset_pin: # CONNECTED TO GND
      pca9554: p_c_a
      number: 5
    de_pin: 18
    hsync_pin: 16
    vsync_pin: 17
    pclk_pin: 21
    init_sequence: 
      - 1
      - [ 0xE0, 0x1F ]
      - [ 0xCD, 0x00 ]
    pclk_frequency: 12MHz
    pclk_inverted: false
    data_pins:
      red:
        - 11         # R1
        - 12         # R2
        - 13         # R3
        - 14         # R4
        - 0          # R5
      green:
        - 8          # G0
        - 20         # G1
        - 3          # G2
        - 46         # G3
        - 9          # G4
        - 10         # G5 
      blue:
        - 4          # B1
        - 5          # B2
        - 6          # B3
        - 7          # B4
        - 15         # B5

    lambda: |-
      it.image(0, 0, id(RGB_test));
      it.print(240, 240, id(roboto20), TextAlign::CENTER,"Hello ESPHome!");
```

agillis, 2024-04-15:
@tiimsvk thanks for all your help getting this working but it's not quite right yet. Once I did more testing I noticed the colors are not right. I am having the same issues that they discuss in this thread on the openHASP project.

https://github.com/HASwitchPlate/openHASP/issues/603

tbxMb get's it working on Jan 2 but needs to add a bunch of code.

https://github.com/tbxMb/openHASP/commit/7af48cff375fc0c73ffa10230f0dc45a7c1a2354

Can we get this added to esphome as well?

Gjuju, 2024-04-15:
Thanks a lot for your reply.

Best regards.

clydebarrow, 2024-04-15:
> Can we get this added to esphome as well?

If it's just a change in the init commands, which a quick glance suggests, you can do that in YAML using the `init_sequence` option.

clydebarrow, 2024-04-15:
> ```
> 
> ```
> 
> 
>     
>   
> 
> I don't understand where can i find that lvlg-package.yam file ?

Mine is here: https://gist.github.com/clydebarrow/ef89e9a93bd44771483b9144ae9042a1

agillis, 2024-04-16:
The init sequence is over 100 lines of C code. I'm not sure how to add this to the YAML file. Would it be possible to add the esp32-s3-4848S040 as a supported board in esphome? That way when you selected the board from a YAML file it would already have all this information? Right now we are using a esp32-s3-devkitc-1 board then trying to mod it in YAML to be a esp32-s3-4848S040.

clydebarrow, 2024-04-17:
Although the init code seems long, it differs only two commands from the existing "standard" init code, so just adding a couple of lines to the yaml will be sufficient.

Supporting specific boards in ESPHome is a two-edged sword - on the one hand it's convenient if your board is already there, OTOH it's more work for devs to keep adding new boards, and ultimately it's a whack-a-mole job - there are always new ones appearing. Allowing init overrides in yaml is providing a way to quickly support new boards without a new release of ESPHome. Configs for specific boards can be contributed to https://devices.esphome.io/ once established (that needs a `display` category.)

The downside is that *someone* still needs to work out what needs to be done for a specific board, and I appreciate that is a "dev" level activity, though there's nothing to stop anyone comparing init sequences.

For this board try this:
```
    init_sequence:
      - 1      # select canned init sequence number 1
      - [0xFF,0x77, 0x01, 0x00, 0x00, 0x00,]
      - [0x21]  #IPS
 ```

That's the only difference between the OpenHasp init code and the ESPHome.

clydebarrow, 2024-04-17:
So as it turns out I actually have one of these esp32-s3-4848s040 boards, so was able to test it out.

The red and blue pin numbers used by OpenHasp are wrong, as is the pixel format - the two errors almost cancelled each other out, but the result was still slightly wrong. Here's a properly working config for ESPHome:

```
#-------------------------------------------
# ESPHOME PLATFORM SETTINGS
#-------------------------------------------
esphome:
  name: 480screen
  friendly_name: 480screen
  platformio_options:
    board_build.flash_mode: dio

#-------------------------------------------
# ESP MCU SETTINGS
#-------------------------------------------
esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: esp-idf
    sdkconfig_options:

      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_SPIRAM_FETCH_INSTRUCTIONS: y
      CONFIG_SPIRAM_RODATA: y

psram:
  mode: octal
  speed: 80MHz

logger:
  level: DEBUG
  hardware_uart: uart0

api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

switch:
  - platform: gpio
    pin: 40
    id: relay1
    name: Relay 1

#-------------------------------------------
# PROTOCOLS
#-------------------------------------------
i2c:
  - id: bus_a
    sda: GPIO19
    scl: GPIO45
    #frequency: 100kHz

spi:
  - id: lcd_spi
    clk_pin: GPIO48
    mosi_pin: GPIO47

#-------------------------------------------
# OUTPUTS
#-------------------------------------------
output:
  - platform: ledc
    pin: GPIO38
    id: backlight
    frequency: 100Hz

#-------------------------------------------
# LIGHTS
#-------------------------------------------
light:
  - platform: monochromatic
    output: backlight
    name: Display Backlight
    id: display_backlight
    restore_mode: ALWAYS_ON

# I/O expander. Touchscreen RESET and INTERRUPT are not used.

touchscreen:
  platform: gt911
  #rotation: 180
  transform:
    mirror_x: false
    mirror_y: false
  id: my_touchscreen
  display: st7701_display

  on_touch:
    - logger.log:
          format: Touch at (%d, %d)
          args: [touch.x, touch.y]
    - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
              touch.x,
              touch.y,
              touch.x_raw,
              touch.y_raw
              );
image:
  - file: https://esphome.io/_images/logo.png
    id: esphome_logo
    resize: 200x162
  - file: cat.jpg
    id: RGB_test
    resize: 480x480
    type: RGB565
   
    
font:
  - file: "gfonts://Roboto"
    id: roboto20
    size: 20
    extras:
      - file: 'fonts/materialdesignicons-webfont.ttf'
        glyphs: [
          "\U000F004B",
          "\U0000f0ed",
          "\U000F006E",
          "\U000F012C",
          "\U000F179B",
          "\U000F0748",
          "\U000F1A1B",
          "\U000F02DC",
          "\U000F0A02",
          "\U000F035F",
          "\U000F0156",
          "\U000F0C5F", 
          "\U000f0084",
          "\U000f0091",
          ]

display:
  - platform: st7701s
    id: st7701_display
    update_interval: 10s
    auto_clear_enabled: True
    data_rate: 2MHz
    spi_mode: MODE3
    color_order: RGB
    invert_colors: false
    dimensions:
      width: 480
      height: 480
    transform:
      mirror_x: false
      mirror_y: false
    cs_pin: 39
      # reset not defined 
    de_pin: 18
    hsync_pin: 16
    vsync_pin: 17
    pclk_pin: 21
    init_sequence: 
      - 1
      - [0x3A, 0x50] # 16 bit pixel format.
    pclk_frequency: 12MHz
    pclk_inverted: false
    data_pins:
      red:
        - 0          # R5
        - 11         # R1
        - 12         # R2
        - 13         # R3
        - 14         # R4
      green:
        - 8          # G0
        - 20         # G1
        - 3          # G2
        - 46         # G3
        - 9          # G4
        - 10         # G5 
      blue:
        - 15         # B5
        - 4          # B1
        - 5          # B2
        - 6          # B3
        - 7          # B4
    lambda: !lambda |-
      it.filled_rectangle(0, 0, 40, 40, Color(0x80, 0, 0));
      it.filled_rectangle(0, 60, 40, 40, Color(0x40, 0, 0));
      it.filled_rectangle(0, 120, 40, 40, Color(0x20, 0, 0));
      it.filled_rectangle(0, 180, 40, 40, Color(0x10, 0, 0));
      it.filled_rectangle(0, 240, 40, 40, Color(0x8, 0, 0));

      it.filled_rectangle(40, 0, 40, 40, Color(0, 0x80, 0));
      it.filled_rectangle(40, 60, 40, 40, Color(0, 0x40, 0));
      it.filled_rectangle(40, 120, 40, 40, Color(0, 0x20, 0));
      it.filled_rectangle(40, 180, 40, 40, Color(0, 0x10, 0));
      it.filled_rectangle(40, 240, 40, 40, Color(0, 0x8, 0));
      it.filled_rectangle(40, 300, 40, 40, Color(0, 0x4, 0));

      it.filled_rectangle(80, 0, 40, 40, Color(0, 0, 0x80));
      it.filled_rectangle(80, 60, 40, 40, Color(0, 0, 0x40));
      it.filled_rectangle(80, 120, 40, 40, Color(0, 0, 0x20));
      it.filled_rectangle(80, 180, 40, 40, Color(0, 0, 0x10));
      it.filled_rectangle(80, 240, 40, 40, Color(0, 0, 0x8));

      it.filled_rectangle(00, 360, 40, 40, Color(255, 0, 0));
      it.filled_rectangle(40, 360, 40, 40, Color(0, 255, 0));
      it.filled_rectangle(80, 360, 40, 40, Color(0, 0, 255));
```

rocob, 2024-05-15:
[clydebarrow](https://github.com/clydebarrow)  this configuration is incorrect because with this you have only achieved a color depth of 4 bits for the colors red and blue. The red and blue pins must remain as originally connected (in the correct order). It is only necessary to change the MDT flag to log 0 (command 0xCD). See [https://focuslcds.com/wp-content/uploads/Drivers/ST7701S.pdf](url) on page 276 and page 73. There is one more command to be given before this command. See the patched code section below also with the pins reversed back.

    init_sequence: 
      - 1
      #- [0x3A, 0x50] # 16 bit pixel format. (wrong)
      #- [0x3A, 0x60] # 18 bit pixel format. (this is default configuration. No need to apply)
      - [0xFF, 0x77, 0x01, 0x00, 0x00, 0x10] # CMD2_BKSEL_BK0
      - [0xCD, 0x00] # disable MDT flag
    pclk_frequency: 12MHz
    pclk_inverted: false
    data_pins:
      red:
        - 11         # R1
        - 12         # R2
        - 13         # R3
        - 14         # R4
        - 0          # R5
      green:
        - 8          # G0
        - 20         # G1
        - 3          # G2
        - 46         # G3
        - 9          # G4
        - 10         # G5 
      blue:
        - 4          # B1
        - 5          # B2
        - 6          # B3
        - 7          # B4
        - 15         # B5
    lambda: !lambda |-

fgh

clydebarrow, 2024-05-15:
Excellent! I tested that out and it does indeed work well. Thanks.