esphome/feature-requests #2634: LORA support in the ESPHome

Jaishankar872, 2024-03-04:
**Describe the problem you have/What new integration you would like**

I have already the home automation setup with help of ESPHome and Home Assistant all using the WiFi. But I want extend the coverage outside the Home area where LORA will be helpful. Please add the Lora into ESPHome

**Please describe your use case for this integration and alternatives you've tried:**

   I have purchased the SX1262 SPI based Lora Module and tried both the Custom SPI Device(with RadioLib library) and External Components (https://github.com/PaulSchulz/esphome-lora-sx126x), but it is failed. and also difficult for me to port

**Additional context**

netweaver1970, 2024-03-04:
Have you looked into https://github.com/StuartsProjects/SX12XX-LoRa 
It's a good option too

Jaishankar872, 2024-03-05:
Thanks for sharing the details

I am using the NodeMCU(ESP8266) on one side and Raspberry pi Pico W on another side with SX1262 lora module 

Currently I am facing lot of struggle on library porting into the ESPHome environment.  

Adding direct support in ESPHome or alternate method will help here.

herostrat, 2024-10-17:
For sx127x modules there is this PR: https://github.com/esphome/feature-requests/issues/2634

swoboda1337, 2025-01-08:
If it is just lora p2p support I could possibly add it to sx127x and https://github.com/esphome/esphome/pull/7490

The sx126x is a very different chip though would be more work.

nagyrobi, 2025-02-07:
Making progress in https://github.com/esphome/esphome/pull/8187, and first implementation is for UART: https://github.com/esphome/esphome/pull/8214. This is a generalised approach, which will make possible to develop simpler components for exchanging data directly between nodes by any means, like TCP, ESP-Now, LoRa, etc.

swoboda1337, 2025-02-17:
Ok I added LoRa support to my sx127x PR https://github.com/esphome/esphome/pull/7490 

Note this is not full LoRaWAN support just LoRa. You can do P2P though and two ESPHome devices should be able to talk to each other a few kms apart.

swoboda1337, 2025-02-18:
> Please also consider adding support for SX126x Lora devices for P2P

I had a heltec v3 lying around and spent a few hours porting my sx127x component. I was able to get basic rx working sx126x. Will take a bit of work to this cleaned up but it works.

swoboda1337, 2025-02-21:
> Please also consider adding support for SX126x Lora devices for P2P

Ok the basics work and you can send and receive LoRa packets. It is still a WIP and the code need significant cleanup plus testing. If you want to try it:

https://github.com/swoboda1337/sx126x-esphome

Can use the sx127x docs for the most part, there is additional tcxo configs (if your device has a tcxo) and sx1261 to differentiate between sx1261 and sx1262/sx1268

https://deploy-preview-4278--esphome.netlify.app/components/sx127x

dala318, 2025-03-07:
First of all, great to see someone taking on this task and really looking forward to start using this!

I do have a question about future improvements, do you plan to add native sensors, switches etc. once the core is merged?
So the `on_packet` and `send_packet` can be utilized to do this but it will involve a bit of lambdas etc. to resolve the raw-to-logical values. Could as well add some basic component types which natively decode raw data and provide that logic.

Ex.
```yaml
sx127x:
  cs_pin: GPIO18
  rst_pin: GPIO23
  id: my_sx1276
  ...

sensor:
  - platform: sx127x
    sx127x_id: my_sx1276  # usually optional
    name: Outdoor temp
    register_key: 0xA001  # Match to first bytes in raw data
    value_factor: 0.1  # For int to float
    # Could be many more parameters for decoding...
    ...
```

Or would this rather be a separate component with dependency to  this sx127x?

swoboda1337, 2025-03-07:
> Would also be interesting if this could be integrated with the new generic packet_transport, so all esphome traffic can be sent over it. For reference see: [esphome/esphome#8187](https://github.com/esphome/esphome/pull/8187), [esphome/esphome#8214](https://github.com/esphome/esphome/pull/8214)

Yes once the generic packet transport stuff is in I can look at merging that. This would make it easy to setup remote sensors etc over lora etc.

bouttier, 2025-03-07:
I had begun a LoRa component before seeing the work of swoboda1337. I suppose https://github.com/esphome/esphome/pull/7490 will be merged one day so I do not open another PR but I share it here as it might be of interest to someone: https://github.com/bouttier/esphome-sx127x
Main difference is the use of the [sx127x](https://github.com/dernasherbrezon/sx127x) external library which has the great advantage over others libs to let you use your own SPI implementation, and so esphome one.
Other features includes async transmission and ability to receive packet in deep-sleep.
Only LoRa modulation is supported as this stage (but FSK/OOK are supported by sx127x library so this can evolve).

swoboda1337, 2025-03-07:
> I had begun a LoRa component before seeing the work of swoboda1337. I suppose [esphome/esphome#7490](https://github.com/esphome/esphome/pull/7490) will be merged one day so I do not open another PR but I share it here as it might be of interest to someone: https://github.com/bouttier/esphome-sx127x
> Main difference is the use of the [sx127x](https://github.com/dernasherbrezon/sx127x) external library which has the great advantage over others libs to let you use your own SPI implementation, and so esphome one.
> Other features includes async transmission and ability to receive packet in deep-sleep.
> Only LoRa modulation is supported as this stage (but FSK/OOK are supported by sx127x library so this can evolve).

I created this originally for OOK, have a lot of weather stuff. 

Async tx could be added, wouldn't be hard. Not sure about receive in deep sleep, might be able to do that with mine as is? I have never tried sleep on the esp32. 

I am open to changes, can open an issue or submit a PR, hopefully we can get support for these modules merged some day.

bouttier, 2025-03-08:
Receive in deep-sleep is easy: you just have to avoid reset/reconfigure the transceiver. I’m not sure async TX is really useful.
I could start contributing to your implementation in the future, but I find it quite satisfying to use an external lib for registers manipulation: this allows to share sx127x code with more users, and benefits from other features like registers caching, so I’m undecided for now…

swoboda1337, 2025-03-12:
> m too working on a device with esphome based on the heltec wifi lora32 v3, and would like to integrate LoRa communication. Let me know if I could help with debugging and development (I'm a software engineer)
> Thanks!

Can give this a try https://github.com/swoboda1337/sx126x-esphome the example yaml was tested on heltec wifi lora32 v3.

dj-fiorex, 2025-03-22:
hello @swoboda1337 i tried your component and it seems to work very good. i'm trying to create a self-healing mesh with LoRa, where any node could be a simple LoRa node or a LoRa<->WiFi gateway. Do you have some insight on how to do this? thanks

dala318, 2025-03-28:
@swoboda1337, got my devices and tested your https://github.com/esphome/esphome/pull/7490 with success!
Really looking forward to make some good use of this (monitoring water pressure at remote irrigation pump station).

While browsing around to learn more about LoRa and LoRaWAN I stumbled upon this abandoned PR https://github.com/esphome/esphome/pull/1609. Have you seen it? What I like about that attempt is that both sx126x and sx127x radios could be combined under one `lora` component, making higher level usage agnostic to the medium. (Now I personally yet only have sx1276 radios, but for those that have others this could bring some extra value)

Config example from that PR:
```yaml
lora:
  - platform: sx127x
    id: lora_12
    sync_word: 0x34
    di0_pin: GPIO26
    rst_pin: GPIO14
    cs_pin: GPIO18
    band: 868MHZ
```

closed 2025-07-03 by bdraco