esphome/feature-requests #2635: SPI Support for ENS160 Air Quality Sensor

lastfreemail40, 2024-03-05:
**Describe the problem you have/What new integration you would like**

It would be great if SPI Support for the ENS160 Airquality sensor was added. There are many ENS160 sensors which also have an AHT21 Temperature and Humidity sensor on board. 
[image]

Because of the heating element from the ENS160 gas sensor the temperature is always 5 or more degrees higher as the room temperature. So the onboard temperature sensor can not be used for baseline calibration.
There is a library from DFRobot (see link in additional context) which already has SPI Support. Maybe there is a way to integrate the SPI support for the existing ENS160 component in ESPHome.
Because of the SPI connection from the ENS160 sensor you loose the readings from the onboard AHT21. But considering the faulty readings from the onboard AHT21 it is not a disadvantage.

**Please describe your use case for this integration and alternatives you've tried:**

My ENS160+AHT21 airquality sensor was connected via I2C and working fine. I tested it for a few days until I recognized the faulty readings from the onboard AHT21. In my case I ordered a standalone AHT21 sensor, which also connects via I2C. I can not wire both sensors on my one I2C Bus because both AHT21 sensors have the same address 0x38.
I tested the sensors with Arduino IDE and wired the ENS160+AHT21 via SPI to my microcontroller. The standalone AHT21 is wired via I2C Bus and working fine.
So I thought it would be possible to add SPI support for the ENS160+AHT21 sensor to ESPHome. 

**Additional context**

[https://github.com/DFRobot/DFRobot_ENS160.git](https://github.com/DFRobot/DFRobot_ENS160.git)
[https://de.aliexpress.com/item/1005004898420675.html](https://de.aliexpress.com/item/1005004898420675.html)

latonita, 2024-03-13:
makes sense. i'll what i can do. i will need you to test it at some point

latonita, 2024-03-13:
@lastfreemail40 that didnt took long.
I need you to test both i2c and spi versions.

```
external_components:
  - source: github://pr#6369
    refresh: 10s
    components:
      - ens160_base, ens160_i2c, ens160_spi

i2c:
  sda: GPIO25
  scl: GPIO32

spi:
  clk_pin: 19
  mosi_pin: 20
  miso_pin: 21

sensor:
  - platform: ens160_i2c
    eco2:
      name: "ENS160 I2C eCO2"
    tvoc:
      name: "ENS160 I2C Total Volatile Organic Compounds"
    aqi:
      name: "ENS160 I2C Air Quality Index"
    update_interval: 60s
    address: 0x53

  - platform: ens160_spi
    eco2:
      name: "ENS160 SPI eCO2"
    tvoc:
      name: "ENS160 SPI Total Volatile Organic Compounds"
    aqi:
      name: "ENS160 SPI Air Quality Index"
    update_interval: 60s
    cs_pin: 22
```

lastfreemail40, 2024-04-26:
Thank you very much. That was really fast :)

lastfreemail40, 2024-04-26:
I was just about to try it with a seeeduino xiao esp32 S3 and got the following error message. there seems to be a missing ___ini__.py

[image]

latonita, 2024-04-27:
What's in the 'source:' section? Can't see behind the popup

image.png (view on web)<https://github.com/esphome/feature-requests/assets/137451969/0832bc35-ec8d-4687-9ad8-c58791087bc3>

latonita, 2024-04-27:
not sure whats happening. i have just updated PR to the latest codebase and compiled successfully with this:
```
external_components:
  - source: github://pr#6369
    refresh: 10s
    components: [ens160_base, ens160_i2c, ens160_spi]
```

lastfreemail40, 2024-04-27:
I tried that, but i missen the brackets. I will test it and give YouTube a Feedback 😀

lastfreemail40, 2024-04-29:
I tried that, but i missen the brackets. I will test it and give YouTube a Feedback 😀

I attached pictures from the log-file. I will test the sensor with the arduino ide to See if it's working. 

[image]

[image]

[image]

lastfreemail40, 2024-04-30:
I checked the sensor yesterday with the arduino IDE and it is working.

latonita, 2024-04-30:
> I checked the sensor yesterday with the arduino IDE and it is working. 

Please send your sketch to Anton.viktorov at live. Com

vincentscode, 2024-05-05:
When trying to test the code from the PR I get the following error message:

```
INFO ESPHome 2024.4.2
INFO Reading configuration /config/esphome/sensor.yaml...
ERROR Unable to import component ens160_i2c.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 169, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/27908350/esphome/components/ens160_i2c/sensor.py", line 3, in <module>
    from ..ens160_base import to_code_base, cv, CONFIG_SCHEMA_BASE
  File "/data/external_components/27908350/esphome/components/ens160_base/__init__.py", line 4, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_COMPENSATION' from 'esphome.const' (/esphome/esphome/const.py)
Failed config
```

I am including the external component as mentioned:
```yaml
external_components:
  - source: github://pr#6369
    refresh: 10s
    components: [ens160_base, ens160_i2c, ens160_spi]
```

Any idea what might cause this? The code changes look good to me, but this keeps me from testing it. :sweat_smile:

latonita, 2024-05-05:
Looks like this is latest (yesterday) change made to core component and the only way to test is to use 'dev' version of esphome

lastfreemail40, 2024-05-15:
I tried to install a second microcontroller and got an error. So I tried to use the dev version of esphome, but I still get an error.
This is how I integrated the external components.

`external_components:
  - source: #github://pr#6369
    type: git
      url: https://github.com/esphome/esphome
      ref: dev
      #refresh: 25s
  components: [ens160_base, ens160_i2c, ens160_spi]'

But I still get an error: 
mapping values are not allowed here
  in "/config/esphome/luftsensornoah.yaml", line 36, column 10

[image]

Could you please help me again...thank you in advance.

closed 2024-05-16 by jesserockz

lastfreemail40, 2024-05-17:
I tried it again yesterday and it worked perfectly.
Thank you 👌