esphome/feature-requests #2636: more waveshare epaper display models support

ELANDJEA, 2024-03-05:
**Describe the problem you have/What new integration you would like**
please add additional waveshare epaper models where possible. For my case the 4.2inch v2 epaper board (https://www.waveshare.com/product/displays/e-paper/epaper-2/4.2inch-e-paper.htm).

**Please describe your use case for this integration and alternatives you've tried:**
when I use the esphome integrated model "2.9inv2" it displays a small part on the panel. other integrated models did not show content on the display (https://esphome.io/components/display/waveshare_epaper.html?highlight=waveshare)

**Additional context**
The demo arduino code by waveshare does have code integrated for almost all models. (https://www.waveshare.com/wiki/4.2inch_e-Paper_Module_Manual#Resources). In theory it must be easy to integrate I think, but do not understand enough of coding to get it done.

JacquesMatthee, 2024-03-17:
Same for the 3.52inch Epaper Display

sheedje, 2024-04-03:
I’d love to see more models supported as well, like the 5.83-inch and the 7.3-inch models. For years, not a single new model has been added besides the standard and never-ending variations of the 2.9, 7.5, 1.54, etc. sizes. While Arduino is blessed with the GXEPD2 library, supporting practically every e-paper screen with full-color support, Esphome users rely on custom components or only black and white support

sheedje, 2024-04-03:
I’d love to see the 5.83 and the 3.7 inch models supported

jazzmonger, 2025-01-31:
Same-  3.52inch Epaper Display