esphome/feature-requests #2639: Please add support for the BH1745NUC Colour Sensor

billmaterial, 2024-03-07:
**Describe the problem you have/What new integration you would like**
The BH1745NUC colour sensor is not yet supported, only the BH1750 is currently.

**Please describe your use case for this integration and alternatives you've tried:**
I have searched for a solution but cannot find anything. I have even tried to adapt the BH1750 code from the esphome repository and  tried make a custom component but I do not have the knowledge to make it work.

**Additional context**

latonita, 2024-03-10:
@billmaterial you may try this one I made today.
Please check if it works at all and post logs via pastebin.
1) RGBC - are raw readings * compensation (2.2f, 1.0f, 1.8f, 10.0f) 
2) Illuminance and Color temp - not sure how good are those, tried to find some algorithms ...
Thanks

```

external_components:
  - source: github://pr#6349
    refresh: 10s
    components:
      - bh1745

i2c:
  sda: GPIO14
  scl: GPIO26
  scan: true
  frequency: 5kHz

sensor:
  - platform: bh1745
    gain: 1X
    update_interval: 20s
    red_channel: Red channel
    green_channel: Green channel
    blue_channel: Blue channel
    clear_channel: Clear channel
    illuminance: Illuminance
    color_temperature: Color temperature
```

billmaterial, 2024-03-11:
Hi @latonita, thank you for your response, just what I was looking for!

I have attempted to write my own and after many hours of work (first time writing a custom component) I managed to get something working.
After looking at your code I realised how bad mine was so I will be using yours from now on :)

I was also trying to add a switch to the frontend for LED control as I have the BH1745NUC breakout board from Pimoroni (https://shop.pimoroni.com/products/bh1745-luminance-and-colour-sensor-breakout?variant=12767599755347)
which has integrated LED's to illuminate the scene. 
I was wondering if you could and if you have time, add this function as an optional switch input (this is the part I struggled on and cannot get working at all yet), I do have it working in my .cpp code to turn it on and off but currently this is hard coded.
The information to do this on this board (if you don't have one of these then it will be difficult to test the functionality but I would do that for you) is basically swap the default values for the threshold registers and enable or disable the Interrupt to turn the LED's on or off like this:

 Set the TH_LSB's and TH_MSB's (0x62 and 0x63) to 0x0000 (Chip Defaults to 0xFFFF)
 Set the TL_LSB's and TL_MSB's (0x64 and 0x65) to 0xFFFF (Chip Defaults to 0x0000)
 
  **uint16_t TH_HIGH[1] = {0xFFFF};
  uint16_t TH_LOW[1] = {0x0000};
  this->write_bytes_16(TH_LSBs, TH_LOW, 1);
  this->write_bytes_16(TL_LSBs, TH_HIGH, 1);**

Then set the INTERRUPT register (0x60) bit INT ENABLE (0) to high (1) to turn the LED on and low (0) to turn it off.

  **// LED control
  //this->set_register_bit(INTERRUPT_I, 0); // turn on LED's
  this->clear_register_bit(INTERRUPT_I, 0); // turn off LED's**

Any help on how to achieve this in home assistant front end would be greatly appreciated.
Thanks' in advance.

latonita, 2024-03-11:
@billmaterial i see. i can add two settings for the thresholds and turning on interrupt.
Is component showing proper readings? 

What about lux/cct calculation - is it close to real life?

billmaterial, 2024-03-11:
@latonita I agree a very crazy idea to use the bh1745's internal N channel FET on the Interrupt line to switch the LED's, but also quite a clever solution so as not to add more chips to the board to control the LED's.

That would be fantastic if those could be added so I can control the LED's from HA's front end as a normal switch.
Doing so would cover everyone's requirements for a normal bh1745 and the Pimoroni's version.
The lux/cct calculation is as far as I can tell correct but I can only compare it to another board (the enviro indoor which uses MQTT) as I have no equipment to calibrate the values.... yet, in fact this is exactly the same code I used in my version to calculate lux and cct.

latonita, 2024-03-12:
@billmaterial 
try new version for the LED
add to yaml
```
switch: 
  - platform: bh1745
    name: BH1754 LED
```

billmaterial, 2024-03-12:
@latonita 
Perfect!!!! Thank you so much for all your support is works flawlessly.

latonita, 2024-03-12:
@billmaterial 
I dont have a device.
Please share couple of different measurements: what were RGBC values, and what were calculated lux and cct.
Need to check some calculations. thanks

billmaterial, 2024-03-13:
[bh1745 data.csv](https://github.com/esphome/feature-requests/files/14584847/bh1745.data.csv)
@latonita 
I have attached a csv file from the HA history, there is not much of a change in the data as most of it is with my office light on or just ambient light from the window (very dark and grey morning at the moment). There are a few spikes when I used the built in LED's to measure close range objects.

If you need more then please let me know what type of readings you need and I will try to record them for you.

latonita, 2024-03-13:
@billmaterial i'm missing clear channel data. can you turn it on?  I only need couple of readings with different light.
Need values for r, g, b, clear and illuminance and cct.
Need to double check it and then will be able to mark PR as 'ready to merge'

billmaterial, 2024-03-13:
@latonita my apologies, I must have accidently removed the clear channel. Here is a 5 minute dump of the data which is the minimum HA will allow.
[bh1745 data.csv](https://github.com/esphome/feature-requests/files/14587579/bh1745.data.csv)

billmaterial, 2024-03-13:
@latonita and another one under different lighting conditions.
[bh1745 more data.csv](https://github.com/esphome/feature-requests/files/14587698/bh1745.more.data.csv)

latonita, 2024-04-07:
@billmaterial can you test last version please, switch is now part of the sensor hub
```
sensor:
  - platform: bh1745
    id: bh1745_sensor1
    address: 0x38
    gain: 1X  # 1X, 2X, 16X
    integration_time: 160ms  # 160ms, 320ms, 640ms, 1280ms, 2560ms, 5120ms
    red_channel: Red channel
    green_channel: Green channel
    blue_channel: Blue channel
    clear_channel: Clear channel
    illuminance: Illuminance
    color_temperature: Color temperature
    update_interval: 20s
    pimoroni_led_switch: LED
```

latonita, 2025-02-20:
@billmaterial i'm not sure it works now due to a lot of changes - i will merge main esphome updates today

closed 2026-10-12 by bdraco