esphome/feature-requests #2641: Make it possible to use two or more Daly BMSs on one ESP

ElektronikCZ, 2024-03-08:
**Describe the problem you have/What new integration you would like**

Its currently only possible to configure one DalyBMS platform on one device. Id like to be able to assign two or more to the hardware UART ports on an ESP32. 
Related to the unsolved community post by a different user: https://community.home-assistant.io/t/esphome-daly-bms-two-devices-on-esp32/451292

**Please describe your use case for this integration and alternatives you've tried:**

I need to monitor 2 Daly BMSs from one device.

**Additional context**

SeByDocKy, 2024-03-31:
ideally shoud be defined as

```
uart:
  - id: uart_0
    tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 9600
  - id: uart_1
    tx_pin: GPIO17
    rx_pin: GPIO16
    baud_rate: 9600

daly_bms:
  - id: bms0
    uart_id: uart_0
    update_interval: 6s
  - id: bms1
    uart_id: uart_1
    update_interval: 6s

sensor:
  - platform: daly_bms
    daly_id: bms0  
    voltage:
      name: "Battery Voltage0"
    current:
      name: "Battery Current0"
    battery_level:
      name: "Battery Level0"
    max_cell_voltage:
      name: "Max Cell Voltage0"
    max_cell_voltage_number:
      name: "Max Cell Voltage Number0"
    min_cell_voltage:
      name: "Min Cell Voltage0"
    min_cell_voltage_number:
      name: "Min Cell Voltage Number0"
 
  - platform: daly_bms
    daly_id: bms1  
    voltage:
      name: "Battery Voltage1"
    current:
      name: "Battery Current1"
    battery_level:
      name: "Battery Level1"
    max_cell_voltage:
      name: "Max Cell Voltage1"
    max_cell_voltage_number:
      name: "Max Cell Voltage Number1"
    min_cell_voltage:
      name: "Min Cell Voltage1"
    min_cell_voltage_number:
      name: "Min Cell Voltage Number1" 

```

FrankM70, 2024-04-09:
This looks nice, when will it work like this?

ElektronikCZ, 2024-04-18:
Whenever someone decides to implement it

latonita, 2024-04-23:
@ElektronikCZ @FrankM70 @SeByDocKy 
try this one https://github.com/esphome/esphome/pull/6615

closed 2024-04-25 by jesserockz