esphome/feature-requests #2646: Support for the zigbee communication protocol via uart in the Tuya component

zoliabraham, 2024-03-16:
**Describe the problem you have/What new integration you would like**

I would like the Tuya component to support tuya's [zigbee communication protocol](https://developer.tuya.com/en/docs/iot/tuya-zigbee-module-uart-communication-protocol?id=K9ear5khsqoty) via uart.

**Please describe your use case for this integration and alternatives you've tried:**

It could be used to replace the chip in Tuya MCU-s and convert them to ESPHome, which communicate via the Zigbee tuya protocol. 
I wanted to replace the original chips in some tuya devices and convert it to ESPHome. The WIFI based ones worked, but I also got some zigbee based ones, in which the MCU uses the zigbee protocol instead for serial communication (linked above). This protocol is not implemented right now in the Tuya component. 
I could not find a viable alternative, except only using the wifi based devices with ESPHome. 

**Additional context**

[This](https://developer.tuya.com/en/docs/iot/tuya-zigbee-module-uart-communication-protocol?id=K9ear5khsqoty) is the device, I wanted to replace the tuya chip in and convert to ESPHome. I only have problem converting the Zigbee based ones.
[Here](https://discord.com/channels/429907082951524364/1217903030943617025) is the thread of the full discord conversation in the ESPHome server. (Mostly the end of it is relevant). @ssieb mentioned, the protocol in the tuya component needs to be specified, since it cannot be detected at runtime. 

This is the relevant part of the uart log output with the zigbee protocol, from which they figured out, it is the zigbee tuya protocol:
[image]

aimwaffel, 2024-04-06:
want to do a similar conversion of a zigbee sensor with tuya zt3l module to an esphome device.
looking at the info you posted it might be enough to enhance the tuya.cpp file - method: 
bool Tuya::validate_message_() {

in this method the communications header is parsed atm following tuya wifi standard.
might actually be a small thing to enhance it to switch between wifi and zigbee tuya protocol.

i must say i did NOT have a look at further commands, if those are similar or very different.

zoliabraham, 2024-04-18:
I am not sure if it would be enough, but I am not very familiar with the codebase. 
I think the logs are very simple, because the esp cannot communicate properly with the MCU, not sure what it would look like with real data.

sebastian-bartkowiak, 2024-06-12:
OK, so I actually wanted to implement same functionality in my own ESPhome setup. I looked at it more closely and the change is not going to be quite as simple as that. There are three major differences between TuYa serial protocol for wifi devices vs TuYa serial protocol for zigbee devices:

1. Frame for zigbee serial protocol includes two additional bytes - between version byte and command byte there are two bytes containing sequence number - `The sequence number is a counter used to keep track of every byte sent outward by a host. It ranges from 0 to 0xFFF0. Once the sequence number reaches 0xFFF0, it loops back to 0.` (extract from TuYa docs - https://developer.tuya.com/en/docs/iot/tuya-zigbee-module-uart-communication-protocol?id=K9ear5khsqoty)

2. Since zigbee is designed to be more battery friendly than wifi there is additional functionality introduced - besides UART lines there are two I/O lines routed between zigbee module and device's MCU - those two lines serve a purpose of waking the devices up from deep sleep modes, when necessary. One line is for the MCU to wake zigbee module up, and the other is for zigbee module to wake the MCU up.

3. (almost) completely different set of commands. The overlap between commands used for wifi module and those used for zigbee module is quite small.

That being said, I started some work on implementing version of the library addressing those differences. I have to say that even though I have experience with programming I have close to no experience in C++, so what I did is almost definitely NOT high-quality-production-grade-software. The code I wrote resides in my fork of esphome repo - https://github.com/sebastian-bartkowiak/esphome_tuya_zigbee - for now I just replaced the code implementing tuya protocol.

The code above is more or less functional in terms of being able to establish some sort of communication with the device. Further work will have to include:
- handling I/O lines mentioned in point 2 above
- actually implementing sequence number counter (for now it is hardcoded to zero)
- finding a way to query the device for all their DPs - theoretically there is command 0x28, which should do just that, but the device that I'm using is not responding to this command at all - will have to investigate on that
- implementing all of the necessary commands (right now only 0x01 - query product information, 0x04 - setting datapoint, 0x06 - proactive DP status are implemented)
- fixing initialization procedure - right now the startup procedure is based only on querying product info
- finally - finding a neat way to merge zigbee functionality back with wifi stack code

I will continue working on that, but no promises that I will actually complete all of the steps mentioned above :)

tjarb, 2026-01-22:
Hi,

So I'm also bumping into the Tuya ceiling mmWave Zigbee sensor when replacing the zigbee TUYA SOC to a ESP8266 (D-mini).
Indeed tried several configs but finally ran into this thread stating different commands and additional IO-lines.

May I ask to the current status?

I'm now stuck with a modified device:

``
[E][tuya:409]: Initialization failed at init_state 1
``

with using the code:

``
substitutions:
  device_ssid: "Tuya Motion Sensor"
  device_name: boiler-motion-sensor
  device_description: "Tuya ZY-M100 Human Prescence Sensor ESP-12F ESP-IDF"
  friendly_name: "Boiler Motion Sensor"
  main_device_id: "boiler-motion-sensor" # Put the name that you want to see in Home Assistant.
  project_name: "tuya.zy-m100-wifi-esp-idf"
  project_version: "1.0"

esphome:
  name: ${device_name}
  comment: ${device_description}
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: "${project_name}"
    version: "${project_version}"

esp8266:
  board: d1_mini

api:
  password: ""

logger:
  baud_rate: 0

web_server:
  port: 80

ota:
  safe_mode: true
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 115200

mqtt:
  broker: 192.168.x.x
  username: ""
  password: ""
  topic_prefix: mmwave

# Register the Tuya MCU connection
tuya:

sensor:
  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${friendly_name} Signal strength
    update_interval: 60s
    internal: true
  # Uptime Sensor
  - platform: uptime
    name: "${friendly_name} Uptime"
    id: uptime_sensor
    update_interval: 360s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? to_string(days) + "d " : "") +
                (hours ? to_string(hours) + "h " : "") +
                (minutes ? to_string(minutes) + "m " : "") +
                (to_string(seconds) + "s")
              ).c_str();
    # Light Sensor
  - platform: tuya
    name: "${friendly_name} Light Intensity"
    id: light_intensity
    sensor_datapoint: 104
    unit_of_measurement: "lux"
    icon: "mdi:brightness-5"
    device_class: "illuminance"
    state_class: "measurement"
    # Distance from Detected Object
  - platform: "tuya"
    name: "${friendly_name} Target Distance"
    id: target_distance
    sensor_datapoint: 9
    unit_of_measurement: "cm"
    icon: "mdi:eye"
    device_class: "distance"
    state_class: "measurement"

text_sensor:
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${friendly_name} IP
    ssid:
      name: ${friendly_name} SSID
    bssid:
      name: ${friendly_name} BSSID
  # Expose Uptime
  - platform: template
    name: ${friendly_name} Uptime Human Readable
    id: uptime_human
    icon: mdi:clock-start

# Restart Buttons
button:
  - platform: restart
    id: "restart_device"
    name: "${friendly_name} Restart"
    entity_category: "diagnostic"
  - platform: safe_mode
    id: "restart_device_safe_mode"
    name: "${friendly_name} Restart (Safe Mode)"
    entity_category: "diagnostic"
number:
    # Sensitivity
  - platform: "tuya"
    name: "${friendly_name} Sensitivity"
    number_datapoint: 2
    min_value: 0
    max_value: 9
    step: 1
    icon: "mdi:ray-vertex"
    # Min Detection Distance
  - platform: "tuya"
    name: "${friendly_name} Near Detection"
    number_datapoint: 3
    min_value: 0
    max_value: 1000
    step: 1
    mode: slider
    unit_of_measurement: "cm"
    icon: "mdi:signal-distance-variant"
    # Max Detection Distance
  - platform: "tuya"
    name: "${friendly_name} Far Detection"
    number_datapoint: 4
    min_value: 0
    max_value: 1000
    step: 1
    mode: slider
    unit_of_measurement: "cm"
    icon: "mdi:signal-distance-variant"
    # Detection Delay
  - platform: "tuya"
    name: "${friendly_name} Detection Delay"
    number_datapoint: 101
    min_value: 0
    max_value: 100
    step: 1
    unit_of_measurement: "s"
    mode: slider
    icon: "mdi:clock"
    # Fading Time - Cool Down Period
  - platform: "tuya"
    name: "${friendly_name} Fading Time"
    number_datapoint: 102
    min_value: 0
    max_value: 1500
    step: 1
    unit_of_measurement: "s"
    mode: slider
    icon: "mdi:clock"

select:
    # Self Check Enum
  - platform: "tuya"
    name: "${friendly_name} Self Check Result"
    icon: mdi:eye
    enum_datapoint: 6
    options:
      0: Checking
      1: Check Success
      2: Check Failure
      3: Others
      4: Comm Fault
      5: Radar Fault

binary_sensor:
    # Status
  - platform: status
    name: "${friendly_name} Status"
    # Occupancy Binary Sensor
  - platform: "tuya"
    name: "${friendly_name} Presence State"
    sensor_datapoint: 1
    device_class: occupancy

``