esphome/feature-requests #2647: Add TSC2007 Touchscreen Controller

MistyMoose, 2024-03-17:
**Describe the problem you have/What new integration you would like**
Add the TSC2007 touchscreen controller. This is found in the new Adafruit TFT Featherwings, and also sold separately to interface with many touchscreens. It communicates via I2C.

**Please describe your use case for this integration and alternatives you've tried:**
This touchscreen controller is integrated into TFT featherwings, the display component of which is already integrated into ESPHome. The touchscreen component can't be currently utilized but is a very useful feature.

**Additional context**
Link to Arduino library: https://github.com/adafruit/Adafruit_TSC2007

Link to tsc2007 data sheet: https://www.ti.com/lit/ds/symlink/tsc2007.pdf?ts=1710686547616&ref_url=https%253A%252F%252Fwww.google.com%252F

Link to product: https://www.adafruit.com/product/5423 and https://www.adafruit.com/product/3315

rgregg, 2025-04-15:
@MistyMoose did you find a solution to this? I'm hoping someone has integrated the TSO2007 already, but if not I'll see if I can cobble something together.

rgregg, 2025-04-20:
I created my own external component here: https://github.com/rgregg/esphome-components/blob/main/components/tsc2007/README.md