esphome/feature-requests #2648: Adafruit MSA301/MSA311 Accelerometer

lboue, 2024-03-18:
**Describe the problem you have/What new integration you would like**

# Adafruit MSA301/MSA311 Accelerometer
### Description

The [MSA311](http://www.adafruit.com/products/5309) is a super small and low-cost triple-axis accelerometer. It's inexpensive, but has just about every 'extra' you'd want in an accelerometer:

* Three-axis sensing, 14-bit resolution
* ±2g/±4g/±8g/±16g selectable scaling
* I2C interface on fixed I2C address 0x62
* Interrupt output
* Multiple data rate options 1 Hz to 500 Hz
* As low as 2uA current draw in low power mode (just the chip itself, not including any supporting circuitry)
* Tap, Double-tap, orientation & freefall detection

The [MSA311](http://www.adafruit.com/products/5309) is very similar to the [MSA301](https://www.adafruit.com/product/4344), but is not a drop-in replacement. The I2C address has changed from 0x26 to 0x62!

### Other

 * [Adafruit MSA311 Triple Axis Accelerometer](http://www.adafruit.com/products/5309)

**Please describe your use case for this integration and alternatives you've tried:**
### Arduino library

 * [Adafruit_MSA301](https://github.com/adafruit/Adafruit_MSA301)

### CircuitPython library

 * [Add MSA311 support](https://github.com/adafruit/Adafruit_CircuitPython_MSA301/pull/18)

**Additional context**

latonita, 2024-05-21:
@lboue I'm on it

labricolenumerique, 2024-05-21:
Nice @latonita

lboue, 2024-05-21:
Thanks @latonita

latonita, 2024-05-22:
@lboue some basic support added. 
few configuration options exposed.

lboue, 2024-05-23:
Thank @latonita . The component built correctly.

latonita, 2024-05-24:
@lboue welcome to check documentation https://github.com/esphome/esphome-docs/pull/3865

baffles, 2024-12-11:
Why was this never merged? Is there any plan to merge?

closed 2025-02-27 by clydebarrow