esphome/feature-requests #2649: Authenticated http Request to Shelly 1 Plus

Sniptzler, 2024-03-18:
**Describe the problem you have/What new integration you would like**

I want to directly communicate with a Shelly Plus 1 via an http request with authentication enabled. For that to work i need a way to authenticate via digest wich is currently not supported by the http_request component.

**Please describe your use case for this integration and alternatives you've tried:**

I thought about constructing the authentication message myself but couldn't find a way to encode in sha256. A way of doing that would resolve my issue. Is there a way i have overlooked?

**Additional context**

The authentication process is described [here](https://shelly-api-docs.shelly.cloud/gen2/General/Authentication).

nagyrobi, 2024-03-18:
Try like this: https://esphome.io/cookbook/http_request_sensor#id2
Not guaranteed to work though, as the Shelly has its own webserver implementation, but who knows...

Sniptzler, 2024-03-18:
I have seen this but to my understanding this always sends the same authorization header in the http request which works in the case of the esphome webserver but not normaly because shelly includes the 
> [nonce:](https://shelly-api-docs.shelly.cloud/gen2/General/Authentication/#successful-request-with-authentication-details) number, random or pseudo-random number to prevent replay attacks, taken from the error message. Required

To my understanding this results in an always changing hash value. But this is only my limited understanding. Correct me if I'm wrong.

nagyrobi, 2024-03-18:
You're understanding correctly.

Sniptzler, 2024-03-18:
Do you think it is possible to get the SHA256 algorythm to work in esphome on the esp8266? Mybe there are even other use cases for this?

nagyrobi, 2024-03-18:
https://wokwi.com/projects/328395940175020627