esphome/feature-requests #2650: Add PCA9698 IO Expander compatibility

Thieuzor, 2024-03-19:
**Describe the problem you have/What new integration you would like**
No support for [NXP PCA9698](https://www.nxp.com/docs/en/data-sheet/PCA9698.pdf) in the main branch.

**Please describe your use case for this integration and alternatives you've tried:**
The PCA9698 is used on a custom made [home automation device](https://github.com/AMBRACE-BV/CiiiXo). Currently it doesn't have support for ESPHome.
Additionally, since there are 80 inputs on the device, it should work through interrupts, which is a fixed GPIO on the ESP32.

**Additional context**
Link to Arduino library: https://github.com/Justin-Pl/PCA9698

mahnunchik, 2024-06-29:
+1 

[image]

roma777roma, 2024-12-07:
news?