esphome/feature-requests #2651: Setting domain for multiple wifi networks

alexiri, 2024-03-20:
**Describe the problem you have/What new integration you would like**
When configuring multiple wifi networks for a device, we should also be able to configure different domains. Each network may have a different domain and it makes little sense that it can't be set per network. Essentially, I'd like to be able to do this:

```
wifi:
  networks:
  - ssid: FirstNetworkToConnectTo
    password: VerySafePassword
    domain: .home
  - ssid: SecondNetworkToConnectTo
    password: VerySafePassword
    domain: .work.com
```

**Please describe your use case for this integration and alternatives you've tried:**
Right now you've got to pick a single domain for all networks and just assume that the device won't work correctly in the others.

**Additional context**

randybb, 2024-03-20:
The domain is just for the dashboard, so instead of `<HOSTNAME>.local.` it will use `<HOSTNAME>.<domain>`. There is no failover mechanism, thus it doesn't make sense.

alexiri, 2024-03-20:
Ok, I see. In my case I have the dashboard running on my laptop and I need to be able to manage a device on two different networks. (The laptop and the device will have to be in the same network for this, of course). It would be really useful if I could specify multiple domains and the dashboard would try each one to see if the device is online.