esphome/feature-requests #2658: Sensor round filter should also support negative precision to round by tens, hundreds, etc.

j9brown, 2024-03-23:
**Describe the problem you have/What new integration you would like**

I'd like to round a value to the nearest ten or hundred.  One way to accomplish this would be to create a round filter with a negative precision such as `round: -1` or `round: -2`.

This representation is consistent with a common understanding of decimal places.  Moreover, everything in the `RoundFilter::new_value` function would work fine with negative precisions.  We just need to change the precision field to allow signed values (currently it is a uint8_t).

**Please describe your use case for this integration and alternatives you've tried:**

I am using `round: 0` for now.  I suppose I could use a sequence of filters to multiply by 0.1 then round then multiply by 10 but that feels unnecessarily cumbersome.

**Additional context**

I'm happy to submit a patch if this feature request is accepted.

gcoan, 2025-09-04:
would round_to_multiple_of not do what you want? 

https://esphome.io/components/sensor/index.html#round_to_multiple_of

j9brown, 2025-09-04:
Yes I think that would work well.  Looks like `round_to_multple_of` was implemented several months after I filed this feature request: https://github.com/esphome/esphome/pull/7142

Allowing negative decimal places might still be convenient.