esphome/feature-requests #2661: ESPHome should not error when `allow_other_uses` is set and pin is not used anywhere else

tjhorner, 2024-03-25:
**Describe the problem you have/What new integration you would like**

Starting in ESPHome 2023.12, ESPHome introduced a validation-time check to ensure that pins used in multiple places are explicitly marked as such, preventing unintended consequences of having multiple components use a pin. It will throw an error if a pin is used in multiple places but not marked with `allow_other_uses`; however, the inverse is also true, and I don't think this should be the case.

**Please describe your use case for this integration and alternatives you've tried:**

As a maintainer of ESPHome packages, I will want to give my users the option to reuse pins however they want in their own config. So for components unlikely to cause issues when used in multiple places I will add `allow_other_uses: true`. However, if the pin is not used in any additional places in the user's own config, ESPHome will throw an error and not let the user proceed. (Example: https://github.com/tjhorner/esphome-standing-desk/issues/7)

Additionally, if a user wants to include a package where `allow_other_uses: true` is not set on a pin they want to reuse, there is no way to do so other than forking and modifying the package to include this.

I don't think it makes sense for ESPHome to outright error during validation if a pin has `allow_other_uses` set but is not used anywhere else; a warning would suffice in my opinion. Users should also be able to disable this check on pins either conditionally (e.g., with a comment) or entirely (e.g., with a command-line flag or as an option in the `esphome:` component).

I understand the implications of using a pin in multiple components and I appreciate the goal of this validation check, but it is not very ergonomic to use without an option to disable or reduce the severity of it, especially when using packages.

**Additional context**

jldesignseu, 2024-03-28:
I kind of have a same request but in diferent scenario. 

I have shift registeres for input and outputs (74hc165 and 74hc595) sharing pins between each other, so that on 4-5 gpio pins i can have 8x n times inputs/outputs. 

In my case now when i describe my IO pin and I use CONF_NUMBER for PIN_SCHEMA I get an error that the pin "is used in multiple places." And when I modify my __init__.py with cv.Optional("allow_other_uses", default=True): cv.boolean then I get "incorrectly sets allow_other_uses: true." for every IO thatn is not used more then once. 

Only solution for me is to go trough the pins and manually mark them as allow_other_uses but only for those that are used more then once.

It would be nice to mark them in schema as **possibly_multiply_use** so it won't generate errors and warnings. 

```
combo74shreg:
  - id: 'combo74shreg_hub'
    data_pin: 16
    clock_pin: 15
    latch_pin: 14
    load_pin: 0
    oe_pin: 13
    sr_count: 4
binary_sensor:
  - platform: gpio
    name: ${node_name}_input_01
    pin:
      combo74shreg: combo74shreg_hub
      number: 0
      mode: input
switch:
  - platform: gpio
    name: ${node_name}_output_01
    pin:
      combo74shreg: combo74shreg_hub
      number: 0
      mode: output
      inverted: False
```
[image]

ssieb, 2024-03-28:
You should never include that option in your component directly.  Make a distinction between the input and output pins.