esphome/feature-requests #2663: Download an ELF file

roboraptor, 2024-03-26:
**Describe the problem you have/What new integration you would like**
Possibility to download ELF files.

**Please describe your use case for this integration and alternatives you've tried:**
Sometimes there is a problem with code and device creates log with section for Exception Decoder.
Instead of going through building ESPhome somewhere else and going through all the other problematics.
It would be nice to have button to download ELF file right from the web UI.
Like somewhere here: 
[image]
If you can download yaml file, you can dowload bin file after manual build, you should be also able to download elf file.

**Additional context**
ESPHome Running on HomeAssistant
Started to get some exception errors after updates without any code change and want to investigate.
Thank you very much <3

roboraptor, 2024-03-26:
https://community.home-assistant.io/t/access-esphome-file-system-to-retrieve-elf-build-file/548973/5

This helped but the button would be better.

yousaf465, 2025-01-06:
I do see this feature in 2024.12.2 
[image]

closed 2025-01-06 by kbx81