esphome/feature-requests #2665: esp-idf support BME680 BSEC

milandzuris, 2024-03-27:
**Describe the problem you have/What new integration you would like**

Please add support for BME860 BSEC esp-idf, because of the framework: arduino i cant run BLE there because it takes up a lot of space.

https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/Implementing-BSEC-2-4-0-0-on-ESP-IDF-5-0-with-ESP32-S1-and-S2/m-p/74787

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

milandzuris, 2024-03-27:
https://github.com/esphome/feature-requests/issues/2431

dgrnbrg, 2024-09-03:
This would be really helpful. I would be happy to make this contribution. Could anyone mentor me through adding esp-idf support for this?

PatrikZuffa, 2025-06-18:
For those who would like to try an alternative, use the BME280 with the ESP-IDF. For me it worked. I can use the BT proxy and the Temperature/Humidity/Pressure sensor. At this point I'm testing the stability of the firmware. I use only one BT device (Govee H5179, the ESP32-WROOM-02D based board).