esphome/feature-requests #2666: TAS5805M support

HA-TB303, 2024-03-28:
**Describe the problem you have/What new integration you would like**

Support for the TAS5805M DAC

**Please describe your use case for this integration and alternatives you've tried:**

The ability to configure and use the TAS5805M as media_player.

**Additional context**

https://www.ti.com/product/TAS5805M

HA-TB303, 2024-04-07:
Managed to get it working with a very basic component largely thanks to ChatGPT ;)
it would be great if ESPHome would support this properly.
https://github.com/HA-TB303/ESPHome-Louder-ESP32

Hedda, 2024-08-19:
FYI, stumbled on this new "***voice-kit***" GitHub repository where ESPHome and Home Home Assistant developers from Nabu Casa are developing several new or improved/enhanced audio and media related components, including extended I2S audio for i2c audio microphone and  i2s audio speaker support, including using audio processor (audio DSP hardware) as well as a new native media player with support for FLAC, WAV/PCM, and more advanced resampling, etc. for the upcoming voicekit hardware reference project from Nabu Casa. Maybe they could add DAC (and ADC) configuration to that project too?

- https://github.com/esphome/voice-kit/
  - https://github.com/esphome/voice-kit/tree/dev/esphome/components
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/nabu
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/media_player.py
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/nabu_media_player.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_pipeline.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_reader.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_decoder.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/resampler.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_mixer.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_resampler.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/flac_decoder.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/wav_decoder.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/mp3_decoder.cpp
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio
      - https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio/speaker
      - https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio/microphone
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/media_player
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/voice_kit
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/audio_audio_processor
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/voice_assistant
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/speaker
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/nabu_microphone
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/microphone

Note that I'm only paraphrasing but one of the representatives that is working on that "***voice-kit***"project more or less wrote there that while ESPHome and Home Assistant voice developers from Nabu Casa are now focusing to work on voice assistant features they are still figuring all this around making use of external audio processors and while they are currently only testing the XMOS xCORE chip as a candidate for an ESPHome-based voice-kit reference hardware design for official Home Assistant Voice Assistant development kit they also plan to work on "audio processor" component for ESPHome with hardware-independent architecture that will not be reliant on specific hardware configurations or dependent specifically on the XMOS xCORE DSP chip but instead allow others to add support for additional DSPs as audio processors (i.e. sound co-processors) in the future, (plus the fact that they will make it so that all the I2S settings and pins are still configurable in YAML, meaning that it should at least be possible add support for DSP types to the  "audio processor" component if they work similar to XMOS xCORE DSP chips, as well as different board designs that uses other I2S settings and pins).  That representative also wrote; "_**we will add all the code to the base ESPHome project once things are stable and working well**_".m and noted that ESPHome and Home Assistant / Nabu Casa developers are right now moving very fast and breaking things as they go so working on code for the new voice-kit related components for ESPHome in a separate repository on GitHub here: 

* - https://github.com/esphome/voice-kit/

Anyway, they already added several new experimental audio features and functions or improvements/enhancements to ESPHome.

Again, still very early in development so addtional components and more development or changes will sure to come there soon.

Suggest at least keep an eye on all related experimental components in the esphome organization's new voice-kit repository:

https://github.com/esphome/voice-kit

https://github.com/esphome/voice-kit/tree/dev/esphome/components

https://github.com/esphome/voice-kit/pulls?q=is%3Apr+

https://github.com/esphome/voice-kit/issues?q=is%3Aissue+

mrtoy-me, 2024-12-15:
Are you still interested in getting a proper component for TAS5805M DAC - I should be able to convert your custom component. I just orders a Louder sound o I will be able to test but if I get time and you want to test, I can try converting before I receive mine

HA-TB303, 2024-12-15:
Yes, I sure am. I can test of you want.

DaveR2408, 2024-12-22:
@mrtoy-me your work is much appreciated! In case you've got someting to test I'm also more than open to help. I bought a S3 version so in case possible a component that works with standard and "S" derivate would be awesome!

mrtoy-me, 2024-12-23:
I bought a esp32-S louder as well and no reason I can think of why both would not work but I have not used an esp32-s yet. I waiting for mine to be delivered so I can do testing - but I might get impatient and publish to github so I can see if I am right track. I will let you know when I do.

HA-TB303, 2024-12-23:
If converted to a proper component, if should also compile on ESP-IDF framework and that's exactly where I'm failing. If it would be easy, I would have done it myself ;)

mrtoy-me, 2024-12-23:
At the moment, I understand official esphome i2s media player component relies on arduino libraries and therefore does not work on idf.

HA-TB303, 2024-12-23:
I have the meda_player component working with ISP-IDF using the adf_pipeline, that is not the issue.

Hedda, 2024-12-30:
> I have the meda_player component working with ISP-IDF using the adf_pipeline, that is not the issue.

FYI, the new nabu media player (which they plan to soon merge upstream to mainline) does not use the adf_pipeline, see:

- https://github.com/esphome/home-assistant-voice-pe/tree/dev/esphome/components/nabu

Hedda, 2025-09-15:
@mrtoy-me see you now made custom TAS5805M component for ESPHome compatible with ESP32-S3 Louder and ESP32 Louder.

* https://github.com/mrtoy-me/esphome-tas5805m

Are you planning to submit that TAS5805M custom component as a pull request to upstream ESPHome to try make it into mainline?

At least the Louder ESP32 development board looks promesing.

Also compatible with the "Louder ESParagus" hardware appliance (which I assume uses the same board in just a nice enclosure?):

* https://www.crowdsupply.com/sonocotta/esparagus-media-center

  * https://www.crowdsupply.com/sonocotta/esparagus-media-center/updates/esphome-driver-changes-to-support-home-assistant-integration-of-louder

I understand that in turn is based on Sonocotta's ESP32 TAS5805M DAC Library?

* https://github.com/sonocotta/esp32-tas5805m-dac/#digital-volume-and-analog-gain

* https://github.com/sonocotta/esp32-tas5805m-dac/#mixer-controls

* https://github.com/sonocotta/esp32-audio-dock

  * https://github.com/sonocotta/esp32-audio-dock?tab=readme-ov-file#using-esp32-audio-boards-with-the-home-assistant

* https://www.seeedstudio.com/Louder-ESP32-p-6401.html

* https://sonocotta.com/louder-esp32/

* https://www.elecrow.com/louder-esp32.html?srsltid=AfmBOoqC_TDAgUdqWXIKgsJVOFm2IN2DHQIYncJmLY1A3yfO2CeRD7rM

PS: Interested in TAS5805M support on Louder ESP32 board in order to act in an experimental "Resonate" client media player setup:

* https://github.com/esphome/esphome/pull/10183