esphome/feature-requests #2667: FDC2112 Support

Tuxtom007, 2024-03-31:
**Describe the problem you have/What new integration you would like**
This sensor is part of a new Soil moisture sensor " SoMo1" from ELV with I2C bus connection.
An SHT40 Humidity and Temperature Sensor is also installed, but this is already supported by ESPHome and I can read it out.
However, it is not possible to read out the actual soil moisture values because the built-in FDC2112 is not supported, but is recognised during the I2C bus scan

**Please describe your use case for this integration and alternatives you've tried:**
Application: Soil moisture measurement for the garden or balcony for automatic irrigation control.
Alternative sensors such as an SMT100 are more expensive.

**Additional context**
[Datasheet fdc2114](https://www.ti.com/lit/ds/symlink/fdc2114.pdf?ts=1711892229896&ref_url=https%253A%252F%252Fwww.google.com%252F)

closed 2024-06-05 by Tuxtom007

piotek, 2024-06-24:
@Tuxtom007 
Du hast das Thema als abgeschlossen markiert ... Aber ist das gelöst worden irgendwie ?
Der FDC2112 ist doch weiterhin nicht im ESPHome und auch ich habe diesen nun auf dem Tisch 
und selbiges Problem ...
Gruss

piotek, 2024-06-24:
danke  ... schade ....
Gruss