esphome/feature-requests #2670: Add support for daikin64 ir remote codes

SodaWithoutSparkles, 2024-04-01:
**Describe the problem you have/What new integration you would like**

Support daikin64 platform in climate

**Please describe your use case for this integration and alternatives you've tried:**
This supports air conditioners that uses the daikin64 remote codes. For example, remote model `R04089036296A`. Arduino heatpump ir library did not include this, but the `IRRemoteESP8266` library supports it (https://github.com/crankyoldgit/IRremoteESP8266/issues/1064).

**Additional context**

Tasmota already supports this, in fact, a lot more codecs. Including: 

```
LG, COOLIX, DAIKIN, KELVINATOR, MITSUBISHI_AC, GREE, ARGO, TROTEC, TOSHIBA_AC, FUJITSU_AC, MIDEA, HAIER_AC, HITACHI_AC, 
HITACHI_AC1, HAIER_AC_YRW02, WHIRLPOOL_AC, SAMSUNG_AC, ELECTRA_AC, PANASONIC_AC, LG2, DAIKIN2, VESTEL_AC, TECO, 
TCL112AC, MITSUBISHI_HEAVY_88, MITSUBISHI_HEAVY_152, DAIKIN216, SHARP_AC, GOODWEATHER, DAIKIN160, NEOCLIMA, 
DAIKIN176, DAIKIN128, AMCOR, DAIKIN152, MITSUBISHI136, MITSUBISHI112, HITACHI_AC424, DAIKIN64, AIRWELL, DELONGHI_AC, 
CARRIER_AC64, HITACHI_AC344, CORONA_AC, SANYO_AC, VOLTAS, TRANSCOLD, TECHNIBEL_AC, MIRAGE, PANASONIC_AC32, 
ECOCLIM, TRUMA, HAIER_AC176, TEKNOPOINT, KELON, TROTEC_3550, SANYO_AC88, RHOSS, AIRTON, HITACHI_AC264, 
HITACHI_AC296, HAIER_AC160, BOSCH144, YORK
```

SodaWithoutSparkles, 2024-04-01:
Also, if possible, I found that the decoded signals of my remote seems to be reversed. Sending the state `ON` would result in the air conditioner **OFF**, and vice versa.

orthobots, 2024-04-17:
am facing the same issue as well. would appreciate integration

SodaWithoutSparkles, 2024-04-30:
I believe that it can be achieved by simply replacing the class and using about the same code...

ferbulous, 2024-05-01:
> Also, if possible, I found that the decoded signals of my remote seems to be reversed. Sending the state `ON` would result in the air conditioner **OFF**, and vice versa. Is it possible to "invert the state"?
> 
> EDIT: After hours of confusion, I found out that the `power` is a toggle field. This means that, if `power` is `True`/`On`, the air-con would invert the current state. Please consider adding a way to fix the wrong toggled state, for example, a lambda call to change the internal power state w/o sending anything.

Were you able to rectify this on tasmota-ir?

SodaWithoutSparkles, 2024-05-01:
> Were you able to rectify this on tasmota-ir?

Tasmota was only used as a proof-of-concept for me to verify that my AC is indeed using `daikin64` and the ir modules are indeed working.

SodaWithoutSparkles, 2024-05-03:
I've been tired of waiting and wrote the code myself. https://github.com/SodaWithoutSparkles/esphome-daikin64

Unfortunately, it seems that, either my code is faulty or esphome is faulty. I cannot get it to compile on a esp32 with esp-idf framework, but the same code on a esp8266 with arduino framework works fine. When compiling on the esp32, it said it cannot find the library despite installing it a few lines before.

orthobots, 2024-05-04:
> I've been tired of waiting and wrote the code myself. https://github.com/SodaWithoutSparkles/esphome-daikin64
>
> Unfortunately, it seems that, either my code is faulty or esphome is faulty. I cannot [...]

Wow. Thanks a lot for recompiling the code for ESP. Will give it a go, as i'm also stuck in the same boat